riscv/riscv-isa-manual #144: LA pseudo instruction implementation

andrewt0301, 2018-03-15:
The load address pseudo instruction (`la rd, symbol`) is described in the reference manual as follows:

`auipc rd, symbol[31:12]`
`addi rd, rd, symbol[11:0]`

In some situations, this implementation seem to cause incorrect results.
The `addi` instruction sign-extends the immediate value (`symbol[11:0]`). So, when `symbol[11]` equals 1, the immediate value is considered negative.

For example, when symbol=0x0000_0FFF `symbol[11:0]` is turned into 0xFFFF_FFFF. 

How to deal with such situations?
Do I get anything wrong?

aswaterman, 2018-03-15:
You are right, of course.  The syntax in the manual is shorthand, and deliberately neglects both PC relativity and sign extension.

Regardless, we are moving the assembly-programming stuff to a separate document entirely: https://github.com/riscv/riscv-asm-manual

andrewt0301, 2018-03-16:
Thanks. I have read this document too.

Eventually, I found out the trick that the RISC-V assembler does to deal with sign-extension of symbol[11:0]. When symbol[11] equals 1, it adds 1 to symbol[31:12]. This resets the high bits that were set to 1 as a result of sign-extension:

`auipc rd, symbol[31:12] + 1`
`addi rd, rd, symbol[11:0]`

Probably, this is worth mentioning in the documentation.

aswaterman, 2018-03-16:
I'll leave that to the folks writing https://github.com/riscv/riscv-asm-manual

closed 2018-03-16 by aswaterman