riscv/riscv-isa-manual #152: Is JAL a J-Type or U-Type instruction? Manual is contradictory on that.

ZiCog, 2018-04-05:
Is JAL a J-Type or U-Type instruction?

I'm reading "The RISC-V Instruction Set Manual Volume I: User-Level ISA Document Version 2.2" from here: https://riscv.org/specifications/. There is contradictory information regarding the instruction type of the JAL instruction.

In the "Preface to Version 2.0" I read the following:

> "The  JAL  instruction  has  now  moved  to  the  U-Type  format  with  an  explicit  destination register, and the J instruction has been dropped being replaced by JAL withrd=x0.  This removes the only instruction with an implicit destination register and removes the J-Type instruction format from the base ISA. There is an accompanying reduction in JAL reach, but a signicant reduction in base ISA complexity."

However in section "2.5 Control Transfer Instructions" the description of JAL states:

> "The  jump  and  link  (JAL)  instruction  uses  the  J-type  format, ... " 

The accompanying instruction format diagram is that of the J-Type.

In Chapter 19 RV32/64G Instruction Set Listings, the table "RV32I Base Instruction Set" still shows JAL as using the J-Type instruction.   

So, which is it? Is there still a J-Type instruction encoding?

aswaterman, 2018-04-05:
We have improved the descriptions of the instruction formats since v2.0, but for transparency we have left the v2.0 preface unchanged.  So, the new text is correct, and the legacy v2.0 preface is, well, legacy.

ZiCog, 2018-04-05:
OK. I agree one should not be changing the release history as shown in the "Preface to Version 2.0".

Perhaps there should be a similar Preface for version 2.1 and 2.2 or whatever that puts the J-Type back again.

aswaterman, 2018-04-05:
This should've gone in the v2.2 preface, but I guess we don't have a flow for "we forgot to record this change in the preface."  Perhaps we rationalized it as falling into the vague bucket of "Numerous additions and improvements to the commentary sections."

closed 2018-06-01 by aswaterman