riscv/riscv-isa-manual #153: sedeleg of misaligned load/store exceptions

pmundkur, 2018-04-11:
The spec says sedeleg[11:9] should be hardwired to zero, but seems to leave open sedeleg[6,4] for load/store misaligned exceptions.  If these exceptions are invisible to U-mode, they should not be delegatable either, right?

I'm not clear about how this plays out with the N extension, but keeping these invisible/non-delegatable seems preferable even in that case.

Or perhaps I missed where this is clarified in the spec.

aswaterman, 2018-04-12:
Actually, misaligned-AMO exceptions are currently exposed in that they cause SIGBUS.

I agree that it probably doesn't make sense to delegate these exceptions to U-mode, but for some reason I'm hesitant to forbid the possibility.

pmundkur, 2018-04-13:
Ok. Related question about delegation:

The text in Section 3.1.13 (medeleg/mideleg) is confusing for mideleg.  The phrase "trap occurs in X-mode" is clear for traps and medeleg, and makes sense.  But the text makes a tight correspondence of delegation between medeleg and mideleg, and the corresponding implied phrase "interrupt occurs in X-mode" doesn't make sense to me.  e.g MTI could occur when running in S-mode.

So it's not clear if mideleg can delegate MEI, MTI or MSI, or are those bits (11, 7, 3) hardwired to 0?  "Traps never transition from a more-privileged mode to a less-privileged mode" makes sense for traps.  It would be good to clarify if something similar holds for interrupts.

aswaterman, 2018-04-13:
Not sure I understand where the confusion lines.

It is possible for an MTI to occur in S-mode.

Interrupts result in precise traps, so the "traps never transition..." statement is true for interrupts.

pmundkur, 2018-04-13:
I'm confused about the answer to this question (yes/no): can mideleg be used to delegate MTI and MEI to S-mode?

If it can be so used, an MTI during S-mode will be handled in S-mode, while an MTI during M-mode will be handled in M-mode (due to no transition to less-privileged mode).   I just want to ensure this is a correct interpretation.

aswaterman, 2018-04-14:
Delegating MTI/MEI is not currently forbidden (though Spike and all of our implementations hardwire those mideleg bits to 0).

But let's assume the implementation did make those bits writable.  If you delegate an interrupt, it is masked at the delegator privilege level.  So an MTI during U/S-mode will be handled in S-mode, whereas an MTI in M-mode is no longer possible.

To put this another way, the "M" in MTI is just a suggestion for the use case; there's nothing fundamentally M-mode about it, unless the implementation chooses to hardwire mideleg[MTI] to 0.

pmundkur, 2018-04-14:
Ok, thanks, I think that clarifies matters for me.

I suspect the source of my confusion was that, for interrupts, "If you delegate an interrupt, it is masked at the delegator privilege level."  That is not true for traps, correct?  If so, perhaps the spec should clarify this.  As currently worded, medeleg and mideleg are tied very closely together, and mideleg is mainly phrased as analogous to medeleg.  But this difference is crucial.

aswaterman, 2018-04-14:
Right.  Perhaps this is best handled with a small commentary block.

aswaterman, 2018-04-14:
Want to propose something?

pmundkur, 2018-04-17:
Sure.  Before I try that though, a related clarification: when say MTIP is delegated to S-mode via mideleg, it should show up in sip as STIP  (as sip does not have an MTIP field), right?  And similarly for MSIP/MEIP? And similarly for sideleg and uip?

aswaterman, 2018-04-17:
No, there's no crosstalk between these fields.  If MTIP were delegated, it would show up in the MTIP bit in sip.

pmundkur, 2018-04-17:
I find this confusing, since the MTIP bit is currently marked as WIRI
in sip, as are the MSIP/MEIP bits.  Perhaps the sip/sie bitfield
diagram needs amending, or a commentary block?

pmundkur, 2018-04-17:
@aswaterman here's an initial attempt.

aswaterman, 2018-04-18:
I'm glad you raised this issue, because it's becoming clearer to me how this text could be misinterpreted.  See my revised version of what you wrote:

https://github.com/riscv/riscv-isa-manual/pull/158

aswaterman, 2018-04-18:
I thought I had added you to the contributors list a while ago, but it seems I forgot.

I'm actually starting to think we should reduce confusion here by writing that mideleg[MxI] "should" be hardwired to 0, and leave it at that.  (But to be clear, if mideleg[MxI] were set, then sie[MxI] would become writable, and the interrupt would show up in sip[MxI].)

pmundkur, 2018-04-18:
Ok, I agree, that is less confusing and makes sense.

closed 2018-11-21 by aswaterman