riscv/riscv-isa-manual #160: FMV.W.X: slightly ambiguous language for XLEN=64.

ccelio, 2018-04-28:
The manual says in Section 8.7:

> FMV.W.X moves the single-precision value encoded in IEEE 754-2008 standard encoding from the lower 32 bits of integer register rs1 to the floating-point register rd. **The bits are not modified in the transfer, and in particular, the payloads of non-canonical NaNs are preserved.**

This is a tad confusing in light of the NaN boxing. The SP 32-bits are not modified, but the highest 32-bits are set to 1. I.e., the SP NaN is untouched but a DP NaN is created.  There is a clarification provided within the same paragraph for RV64 and `fmv.x.w` (upper 32 bits are sign-extended). I believe we should provide a similar clarification for `fmv.w.x`.

**Recommendations:**

* Add a paragraph break between the `fmv.w.x` and `fms.x.w` discussion.

* Explicitly mention that the upper 32 bits of the integer `rd` should be 1s, to denote a DP NaN as specified in section 9.2 (which comes later in a different chapter of the manual).

aswaterman, 2018-04-28:
This was deliberate, because this chapter only describes FLEN=32.  The expectation is that people would read Sec. 9.2 if they are building implementations with FLEN>32.

That said, I don't have a problem adding the forward reference.  I'll make a PR.

aswaterman, 2018-04-28:
See https://github.com/riscv/riscv-isa-manual/pull/161

aswaterman, 2018-04-28:
I changed my mind; the resulting text was even more confusing than before.  It implied by omission that FMV.W.X was the only interesting special case when FLEN>32.

I'll add the paragraph break, but I'm not sure how to address your main concern in a way that's not misleading.