riscv/riscv-isa-manual #163: URET from U-mode without N extension

gameboo, 2018-05-01:
Hello.
I think the spec is not clear in [3.2.2](https://github.com/riscv/riscv-isa-manual/blob/ba5cfa4d1eb0aeb9e694dbaf6ccb05312cca02b0/src/machine.tex#L1970) about what should happen when a URET is executed from U-mode when the N extension is not present:

> URET is only provided if user-mode traps are supported

I presume that implicitly means "raise an illegal instruction exception"?

benjaminselfridge, 2018-05-01:
Yep, that's right. "Not supported" generally means "is considered an illegal instruction."

gameboo, 2018-05-01:
OK cool. It would be worth being explicit here, no?
Thanks again.

benjaminselfridge, 2018-05-01:
If you like, go ahead and submit a pull request; it has a decent chance of being integrated, I'd bet.

closed 2018-05-23 by gameboo