riscv/riscv-isa-manual #174: unique trap causes for TW/TSR/TVM

michaeljclark, 2018-05-23:
on the basis that:

- we don't make the page fault handler check epc for a load/store
- it is easy now and hard later (TW/TSR/TVM are not in widespread use)
- exit causes might be common for other hypervisor use cases
- we have 2^32 - 15 causes left

I can make a proposed edit if the idea is accepted in-principle.

michaeljclark, 2018-05-23:
not having cause in siginfo on linux is kind of orthogonal, as that is U-mode not HS-mode/S-mode. That said, that would be nice too e.g. regs[REG_SCAUSE]. It was noted on the QEMU list that some other platforms included this info in their signal trap handler context structure. This is orthogonal to this request... just it came to mind... as it is somewhat related... we need to parse the instruction to figure out whether we got a load or store fault for SIGSEGV in user-mode.

aswaterman, 2018-05-23:
We can’t make backwards incompatible changes like this. (Unless it only affects hypervisor mode.)

I don’t see what this has to do with checking epc on ld/st faults, though. TVM etc. cause illegal-instruction exceptions, not page faults.

aswaterman, 2018-05-26:
Ah, my misunderstanding.

Still, I don't think that partially decoding these instructions is too onerous.  Since tval is populated with the instruction word, there's no need to dereference epc, which is the expensive part.

Note also, this isn't the only case where this pattern shows up.  Many machines trap and emulate misaligned loads and stores, hence need to load and decode instructions inside the trap handler.

closed 2018-11-21 by aswaterman