riscv/riscv-isa-manual #175: Nightly PDF?

daniellustig, 2018-05-23:
AFAICT there is no nightly build of these PDFs available anywhere, and that makes it hard for people to easily see what's new since the latest stable version posted at https://riscv.org/specifications.  In particular, a few different people have now asked me for PDFs so that they can see what's in the memory model we proposed, but they don't want to have to install latex and recompile the docs themselves.  Mostly they just ask me to send a PDF instead :)

Is there any way some kind of automatic nightly build could be set up, such that a nightly PDF could appear at https://riscv.org/specifications?  Or something along those lines?  Possibly with the Foundation's help?

Or, if this exists already, can someone point me to it?

Thanks!

seldridge, 2018-05-23:
Seems like building on Travis, publishing to GitHub page, and updating riscv.org links to point at the GitHub pages files would do it?

aswaterman, 2018-05-23:
Yeah, I'm working on that now.  (Except for changing the riscv.org links--not sure if publishing nightly builds there is a good idea, and it's not my call, anyway.)

ilg-ul, 2018-05-23:
> Seems like building on Travis, publishing to GitHub page, and updating riscv.org links to point at the GitHub pages files would do it.

should do it.

here is an example of a travis config calling a support script which publishes on GitHub:

- https://github.com/gnu-mcu-eclipse/gnu-mcu-eclipse.github.io-source/blob/master/.travis.yml
- https://github.com/gnu-mcu-eclipse/gnu-mcu-eclipse.github.io-source/blob/master/scripts/travis.linux.sh

aswaterman, 2018-05-23:
Thanks for the examples.

aswaterman, 2018-05-24:
OK, got it.  See the README.  Thanks @ilg-ul; your example helped me find a bug in my .travis.yml.

ilg-ul, 2018-05-24:
You're welcome!

daniellustig, 2018-05-24:
That was quick!  Thanks all!

closed 2018-05-24 by daniellustig