riscv/riscv-isa-manual #176: SFENCE.VMA and PMP interactions

michaeljclark, 2018-05-25:
"As noted on isa-dev, we forgot to write down the interaction between
PMPs and TLBs.  The idea is that VA->PA mappings can be cached, and
the PMP access can occur any time between the page-table walk and the
access itself.  In operational terms, PMPs can be cached in the TLB.
Hence, SFENCE.VMA should be executed after modifying PMPs that
underpin virtual-address translations." @aswaterman 

No required for SFENCE.VMA if satp.vm = bare.

aswaterman, 2018-06-01:
I believe  https://github.com/riscv/riscv-isa-manual/commit/4a29140ef57e38532b3d3e43c9cd49e07066e7e0 addresses this issue; please reopen if I missed something.

closed 2018-06-01 by aswaterman

jillleon007, 2025-02-07:
"If page-based virtual memory is not implemented, memory accesses check the PMP settings synchronously, so no SFENCE.VMA is needed."

How to understand "not implemented"? for example，if the core support sv39，when the satp.vm=bare，Does it need sfence.vma to sync the pmp new config？

jillleon007, 2025-02-07:
> I believe [4a29140](https://github.com/riscv/riscv-isa-manual/commit/4a29140ef57e38532b3d3e43c9cd49e07066e7e0) addresses this issue; please reopen if I missed something.

"If page-based virtual memory is not implemented, memory accesses check the PMP settings synchronously, so no SFENCE.VMA is needed."

How to understand "not implemented"? for example，if the core support sv39，when the satp.vm=bare，Does it need sfence.vma to sync the pmp new config？

gfavor, 2025-02-07:
Irrespective of what satp.mode one operates in, if sv39 is supported, that equates to "page-based virtual memory is implemented".  And hence software needs to use an sfence.vma to sync PMP changes.

To help this make sense, note that designs that implement page-based virtual memory will have TLBs and will typically cache translations for all modes of operations.  And, as part of caching translations, the associated PMA and PMP information will also be cached.

So, for Bare mode, trivial 1-to-1 address translations, along with associated PMA and PMP info, will be cached.  In essence, Bare mode is not a special case for all the address datapaths.

jillleon007, 2025-02-07:
> Irrespective of what satp.mode one operates in, if sv39 is supported, that equates to "page-based virtual memory is implemented". And hence software needs to use an sfence.vma to sync PMP changes. [...]

Thank you！