riscv/riscv-isa-manual #186: fence.tso and ignored fence fields

asb, 2018-06-07:
The current draft specification introduces fence.tso, which makes use of the previously unused bits in instr[31:28]. Since at least version 2.0, implementers were told to ignore these fields (although this was made much more clear in version 2.1):

> The unused fields in the FENCE instruction, imm[11:8], rs1, and rd, are reserved for finer-grain
> fences in future extensions. For forward compatibility, base implementations shall ignore these
> fields, and standard software shall zero these fields.

I'd suggest that the description of alternate values for imm[31:28] as "invalid" might muddy the waters a bit. I would expect trapping to be a valid behaviour for "invalid" field values and the specification doesn't seem to indicate otherwise. But surely we want implementers to continue to ignore unrecognised fm field values to allow future extensibility?

I would suggest:
1) Changing use of 'Invalid' in table 2.2 to 'Ignored'. This isn't an ideal description as the table tries to simultaneously describe mnemonics as well as encodings, and of course unrecognised mnemonics should result in errors in software.
2) Clarify the intended interpretation of "Likewise, all combinations of fm and predecessor/successor
set not listed in Table 2.2 are also reserved for future use." Again, I assume the intent is that these encodings are ignored by compliant cores. "Reserved for future use" doesn't seem sufficiently clear about whether these encodings should trap or not.

A related comment:
`fence.tso` was added to the table of pseudo-instructions and expands to `fence.tso rw, rw`. Other pred/succ specifications aren't valid (as explained in section 2.9). Given that, is there any point in accepting the `fence.tso rw, rw` form? Wouldn't it be simpler to define `fence.tso` as an instruction and then later specify it as a pseudoinstruction in the case that alternate pred/succ arguments are accepted.

aswaterman, 2018-06-07:
I agree `fence.tso` should take no operands, and this is how I implemented it in binutils.  If fence.tso is later extended to accept things other than rw, rw, then at that time it can be extended to accept operands.

aswaterman, 2018-06-07:
IMO, changing it to "Ignored" isn't quite right, either: it implies to me that software is allowed to put stuff there.  "Ignored" correctly describes what hardware should do, but it remains "invalid" for software to use those encodings.

@daniellustig want to take a stab at clarifying this?

aswaterman, 2018-06-07:
I'll update the pseudoinstruction listing.

asb, 2018-06-07:
Ah, I just submitted #187 for the pseudoinstruction listing. Feel free to close if you've already authored your commit.

Good point about "ignored" also being misleading, I agree. Maybe it's best to remove the comments about unrecognised encodings from the table and to rely on a longer form prose description to give proper clarity.

aswaterman, 2018-07-06:
Resolved by #201

closed 2018-07-06 by aswaterman