riscv/riscv-isa-manual #189: does WFI ever retire?

pmundkur, 2018-06-10:
Could the manual clarify whether a WFI ever retires?  A naive reading would be it never does, since it either traps or is interrupted.

If it never retires, then this line in the privileged spec may be misleading:

"The WFI instruction is just a hint, and a legal implementation is to implement WFI as a NOP."

since a NOP always retires.  Since the effect on minstret is different, is a NOP really a legal implementation of WFI?

gameboo, 2018-06-10:
I was wondering that!

mwachs5, 2018-06-10:
Even a “true” WFI (one implemented as more than a NOP) doesn’t have to be
interrupted to exit. It only relies on MIP bits being nonzero to retire,
not MIP&MIE&MSTATUS.mie, which is needed to actually take the interrupt.

aswaterman, 2018-06-11:
I don't feel there's a need for a clarification here.  The manual already indicates two conditions under which WFI can retire: it can be implemented as a NOP, as @pmundkur mentioned, and it retires when there's an unmasked pending interrupt, as @mwachs5 pointed out.

pmundkur, 2018-06-11:
Got it, thanks for the clarifications.  Closing.

closed 2018-06-11 by pmundkur

scottj97, 2021-01-27:
Followup question. The priv spec says:

> If an enabled interrupt is present or later becomes present while the hart is stalled, the interrupt exception will be taken on the following instruction, i.e., execution resumes in the trap handler and mepc = pc + 4.

If the interrupt exception is taken on the following instruction, doesn't that mean that the WFI has retired? Especially since `mepc` is pointing beyond the WFI.

It appears that Spike does not increment `minstret` in this case, but I don't know why it shouldn't.

aswaterman, 2021-01-27:
I agree if `mepc` points after the WFI, then the WFI has retired.  It's plausible you found a corner-case bug in Spike's handling of `minstret`.

eroom1966, 2023-06-23:
re-awakening this thread I would also like further clarification & confirmation.
The question is not really related to how spike has implemented this behavior,
it is about how the WFI should behave under all possible scenarios.

What needs explicit clarification is whether the stall occurs before or after the WFI retires 
It seems that we have the following possible outcomes

1. Retires WFI as a NOP instruction
**does not stall the processor, and cannot be interrupted**
```
if (fetch(PC) == WFI) {
    retire(PC)
    PC = PC+4
}
```

2. Retires WFI - then stalls the processor, awaits an interrupt. 
If interrupts enabled - traps to the handler,
if interrupts disabled resumes at PC+4
```
if (fetch(PC) == WFI) {
    retire(PC)
    wait(interrupt)
    if (enabled(interrupt)) {
        MEPC = PC+4
        PC = IRQHANDLER
    } else { 
        PC = PC+4
    }
}
```
The important note here, is that the stall occurs **AFTER** the WFI has retired

3. In the case of TW=1 
Stalls for some (implementation specific) period then either 
retires (interrupt pending but disabled), continues at PC+4
retires (interrupt pending and enabled), takes interrupt trap
times out, does not retire, takes an illegal instruction exception
```
if (fetch(PC) == WFI) {
    wait(interrupt || timeout)
    if (interrupt) {
        retire(PC)
        if (enabled(interrupt)) {
            MEPC = PC+4
            PC = PC.IRQHANDLER
        } else { 
            PC = PC+4
        }
    } else if (timeout) {
       MEPC = PC
       PC = PC.ILLEGAL_INSTRUCTION
    }
}
```
The distinction to be made is case (3) versus (1, 2)
In case (3) the processor **must** stall **BEFORE** either retiring the WFI, trapping, or continuing
In case (1, 2) the processor **must** stall **AFTER** retiring the WFI
(**note in case 1, stall time is zero)

Does this seem like a fair summary ?

aswaterman, 2023-06-23:
That sounds right, except note that hybrid schemes are possible.  Consider the following algorithm, which combines cases (1, 2) and (3):

- stall the WFI for a bounded amount of time, or until an interrupt becomes pending
- check whether TW would result in an exception; raise illegal-instruction exception if so
- otherwise, retire the WFI, then stall until an interrupt becomes pending