riscv/riscv-isa-manual #196: Misleading wording in fused multiply-add description

tymcauley, 2018-06-17:
In the F extension section of the User-Level ISA manual, the four fused multiply-add instructions (FMADD.S, FMSUB.S, FNMADD.S, and FNMSUB.S) are described as follows:

> Fused multiply-add instructions multiply the values in _rs1_ and _rs2_, optionally negate the product, then add or subtract the value in _rs3_, writing the final result to _rd_.

However, if you look at the behavior implied by the FMADD riscv-tests repo (https://github.com/riscv/riscv-tests/blob/master/isa/rv64uf/fmadd.S), the description doesn't accurately define the FNMADD and FNMSUB instructions. The optional N character doesn't "optionally negate the product" as the text says (where the product is _rs1_ times _rs2_), it optionally negates the output of the fused multiply-add/subtract.

The relevant code can be found here: https://github.com/riscv/riscv-isa-manual/blob/master/src/f.tex#L447

I would suggest that we change this sentence to something like this:

> Fused multiply-add instructions multiply the values in _rs1_ and _rs2_, add or subtract the value in _rs3_, optionally negate the sum or difference, then write the final result to _rd_.

On a slightly separate note, I think that adding some parentheses to the FMA equations after this sentence might also make things a bit clearer. While the equations are correct with regard to order-of-operations (multiply before add/subtract), adding some parentheses around the multiply expressions might make things a bit clearer. Furthermore, factoring out the negation for the FNMADD and FNMSUB expressions might clarify their behavior as well (since their equations will then be identical to FMADD and FMSUB, just with a negation in front). For example, the original text says:

> FNMADD.S computes -_rs1_*_rs2_-_rs3_

We might instead write:

> FNMADD.S computes -((_rs1_*_rs2_)+_rs3_)

Hopefully this isn't overly-pedantic, but when I was reading through this section, the combination of the "optionally negate the product" comment and the unfactored equations threw me for a bit of a loop. Just want to make this clearer for whoever reads it next.

aswaterman, 2018-06-18:
This is a duplicate of https://github.com/riscv/riscv-isa-manual/issues/103

Yes, it’s misleading, but as discussed in the other issue, the proposed change is also misleading. I suggest continuing the discussion on the aforementioned issue if you have other ideas on how to clear things up.

closed 2018-06-18 by aswaterman

tymcauley, 2018-06-18:
Ah, sorry about that. Failed to find that issue when I searched before. I'll move discussion to there.