riscv/riscv-isa-manual #202: Small issues in RVWMO Explanatory Material

tymcauley, 2018-06-22:
I believe I've found three issues in the RVWMO Explanatory Material chapter ([memory.tex](https://github.com/riscv/riscv-isa-manual/blob/master/src/memory.tex)), but I wanted to run them by you all before submitting a PR. This section is definitely not in my wheelhouse.

This is probably most relevant for @daniellustig.

#### Issue 1: Section A.3.7 (`sec:memory:acqrel`)

The two figures in this section, A.7 (`fig:litmus:spinlock_atomics`) and A.8 (`fig:litmus:spinlock_fences`), demonstrate how to implement a spinlock either with atomics that are annotated with `ar`/`rl`, or by using atomics plus fences. [This line](https://github.com/riscv/riscv-isa-manual/blob/master/src/memory.tex#L539) is drawing a distinction between the two figures, but mistakenly refers to Figure A.8 twice at the very end of the line. I believe that the second reference to `fig:litmus:spinlock_fences` should be changed to a reference to `fig:litmus:spinlock_atomics`.

Proposed diff (scroll all the way to the right to see the change):

```
@@ -539 +539 @@ Release orderings work exactly the same as acquire orderings, just in the opposi
-This ensures, for example, that memory accesses in a critical section appear before the lock-releasing store in the global memory order.  Just as for acquire semantics, release semantics can be enforced using release annotations or with a FENCE~R,RW operations.  Using the same examples, the ordering between the loads and stores in the critical section and the ``Arbitrary unrelated store'' at the end of the code snippet is enforced only by the FENCE~R,RW in Figure~\ref{fig:litmus:spinlock_fences}, not by the {\em rl} in Figure~\ref{fig:litmus:spinlock_fences}.
+This ensures, for example, that memory accesses in a critical section appear before the lock-releasing store in the global memory order.  Just as for acquire semantics, release semantics can be enforced using release annotations or with a FENCE~R,RW operations.  Using the same examples, the ordering between the loads and stores in the critical section and the ``Arbitrary unrelated store'' at the end of the code snippet is enforced only by the FENCE~R,RW in Figure~\ref{fig:litmus:spinlock_fences}, not by the {\em rl} in Figure~\ref{fig:litmus:spinlock_atomics}.
```

#### Issue 2: Section A.3.9 (`sec:memory:ppopipeline`)

[This line](https://github.com/riscv/riscv-isa-manual/blob/master/src/memory.tex#L760) is comparing the example in Figure A.14 (`fig:litmus:addrdatarfi`) to the example in Figure A.15 (`fig:litmus:addrdatarfi_no`). The line is noting the change in behavior between A.14 and A.15 due to the addition of the `sw a0, 0(s2)` instruction in Hart 1 in A.15. I believe the line mistakenly uses the alphabetic indices (a), (b), and (c) when it means to use (d), (e), and (f).

Proposed diff:

```
@@ -760 +760 @@ Therefore, (f) will never perform before (d) has performed.
-If there were another store to the same address in between (b) and (c), as in Figure~\ref{fig:litmus:addrdatarfi_no}, then (c) would no longer be dependent on the data of (b) being resolved, and hence the dependency of (c) on (a), which produces the data for (b), would be broken.
+If there were another store to the same address in between (e) and (f), as in Figure~\ref{fig:litmus:addrdatarfi_no}, then (f) would no longer be dependent on the data of (e) being resolved, and hence the dependency of (f) on (d), which produces the data for (e), would be broken.
```

#### Issue 3: Section A.3.9 (`sec:memory:ppopipeline`)

[This line](https://github.com/riscv/riscv-isa-manual/blob/master/src/memory.tex#L788) is analyzing the example in Figure A.16 (`fig:litmus:addrpo`), and noting that instruction (f) (`sw t1, 0(s0)`) must wait until we know the load address for instruction (e) (`lw a2, 0(a1)`), since we don't want the store to clobber the data loaded in (e). However, this line notes the address comparison as `a0=s1` when it should read `a1=s0`.

Proposed diff:

```
@@ -788 +788 @@ Consider Figure~\ref{fig:litmus:addrpo}:
-(f) cannot be executed until the address for (e) is resolved, because it may turn out that the addresses match; i.e., that {\tt a0=s1}.  Therefore, (f) cannot be sent to memory before (d) has executed and confirmed whether the addresses do indeed overlap.
+(f) cannot be executed until the address for (e) is resolved, because it may turn out that the addresses match; i.e., that {\tt a1=s0}.  Therefore, (f) cannot be sent to memory before (d) has executed and confirmed whether the addresses do indeed overlap.
```

Let me know if those changes look appropriate, and I can make a PR to fix them.

daniellustig, 2018-06-25:
Thanks @tymcauley!  I think you're right on all counts.  Thanks for the careful read and for the corrections.

Since I already have open PR #201 for some other small memory model corrections, let me just apply these changes there to simplify the merge.  I'll credit you and this issue when I do that.

closed 2018-06-25 by aswaterman

tymcauley, 2018-06-25:
Sounds great, thanks for taking care of that @daniellustig!

commit 74573682 (2018-06-25): Correct some memory model explanation typos.

Thanks to @tymcauley.  Closes #202.