riscv/riscv-isa-manual #204: Question on interactions with SFENCE.VMA

gameboo, 2018-06-28:
In the current spec, from what I understand so far:

- writes to `pte`s in memory need to be synchronized with `sfence.vma`
- writes to `pmp` csrs need to be synchronized with `sfence.vma`
- wrties to the `satp` csr **do not** need to be synchronized with `sfence.vma`

I simply want to confirm that this is indeed how the spec should be understood, as it felt slightly odd to me that `pmp` csrs need explicit synchronization and `satp` does not, implying some sort of microarchitectural support (probably a pipeline bubble in the most trivial implementations).

Thanks.

aswaterman, 2018-06-28:
Yes, that's all true.

aswaterman, 2018-06-29:
Mostly to make it easier to cache the PMPs inside cache tags, TLB entries, etc.  satp itself doesn't get cached anywhere.

gameboo, 2018-07-01:
ok. thanks.

gameboo, 2018-07-01:
Just realised I wanted to confirm something else on the same topic:
Say I have valid PTEs in a cache of some sort, and I update the PMP in such a way that I would no longer be allowed to fetch those PTEs. It is valid behaviour to still hit on those PTEs after the PMP update despite the fact that access to the PTEs is now forbidden, and this until an SFENCE.VMA instruction with rs1=x0 and rs2=x0 is executed. Correct?

aswaterman, 2018-07-02:
Yeah that's right.

gameboo, 2018-07-03:
Great thanks.

closed 2018-07-03 by gameboo