riscv/riscv-isa-manual #207: CSR values when changing MXLEN/SXLEN/UXLEN and when observing from M/S/U-mode

gameboo, 2018-07-02:
I was wondering about the value of CSRs when changing the effective XLEN, as well as the relationship between `mstatus`/`sstatus`/`ustatus`... I basically ended up going through #108 and #120 in my head, realizing that what I was after was basically implicitly defined in the document already (basically projection operations). For what it's worth, I think it would have been easier to convince myself that there weren't holes in the spec with the notion of projection being explicit (both for projection between XLENs and projections between privilege modes).
What I am still wondering about though, and I am not sure how much of an irrelevant side track this is, is whether those two scenaios should ever be different scenarios or not:
- writing a CSR in M-mode 32-bit, changing MXLEN to 64, changing to S-mode, reading the CSR
- writing a CSR in M-mode 32-bit, changing to S-mode, changing MXLEN to 64, reading the CSR
In other words, in general, how do the projections commute with each other.

aswaterman, 2018-07-02:
Scenario 2 isn't possible as written, as S-mode can't cause MXLEN to change.  But if it were, the result should be the same.

gameboo, 2018-07-03:
I guess MXLEN was a poor choice. Say it was an SXLEN CSR... I presume this would work out fine, and the expected result would also be the same based on your previous answer ?

gameboo, 2018-07-03:
> I'm not sure what problem you're anticipating.

Probably none whatsoever... I was just fighting with my mental model here:

[image]

*For context, I am working on an implementation, and therefore was wondering about an appropriate language construct (probably a few typeclass methods) to represent this mechanism in a maintainable way, being able to cope with future CSRs, only having to implement a few function and have those being called automatically...*

Writing it down like this made me be worried that some sort of constraint would be needed for `g.f` to be the same as `g'.f'`... But I am realising now that, by vertue of the view of a CSR being defined for a given privilege level and a given XLEN (here view D), we mandate that `g.f` = `g'.f'`

closed 2018-07-03 by gameboo