riscv/riscv-isa-manual #213: "F" and "D" were changed and the version should be bumped

tommythorn, 2018-07-18:
The table in the preface has not received enough attention - for example the floating
point specification changed materially, including this paragraph

```
+Apart from transfer operations described in the previous paragraph,
+all other floating-point operations on narrower $n$-bit operations,
+\mbox{$n<$ FLEN}, check if the input operands are correctly NaN-boxed,
+i.e., all upper FLEN$-n$ bits are 1.  If so, the $n$ least-significant
+bits of the input are used as the input value, otherwise the input
+value is treated as an $n$-bit canonical NaN.
```

An compliant 2.0 implementation prior to this update will not pass the current tests.
Clearly that means that this is an incompatible change and we must bump the version.

I haven't combed the rest of them to see if there are others that need correction.

aswaterman, 2018-07-18:
I think we should've actually bumped the version twice, since the NaN-boxing scheme adopted in v2.2 of the spec was also backwards incompatible (and much more obviously so).

aswaterman, 2018-08-28:
Resolved by eb7817141bbbaa94157d65ef10c33a280dba1434 -- thanks for bringing this up.

closed 2018-08-28 by aswaterman