riscv/riscv-isa-manual #217: Ambiguity of behavior of store-conditional, missing reservations, and exceptions.

ccelio, 2018-08-04:
**Question: What is the expected behavior if a SC is misaligned but there is no load reservation?**

I do not find any definitive description in the ISA manual regarding this behavior.

> For LR and SC [...] If the address is not naturally aligned, a misaligned address exception will be generated.

That suggests the lack of a reservation is immaterial and the misaligned exception should occur.

However, if I look at Spike, the golden model, it looks like it only checks for a MA exception if the reservation is available:

https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/sc_w.h:

```
require_extension('A');
if (MMU.check_load_reservation(RS1))
{
  MMU.store_uint32(RS1, RS2);
  WRITE_RD(0);
}
else
  WRITE_RD(1);

MMU.yield_load_reservation();
```

allenjbaum, 2018-08-05:
I agree that this looks like a bug in Spike; the spec is very clear, and (based on a discussion I had with Chronis last week) there are good reasons for it.

aswaterman, 2018-08-05:
There’s no ambiguity; as Allen said, this is a Spike bug.

aswaterman, 2018-11-06:
Fixed via https://github.com/riscv/riscv-isa-sim/commit/01252686902fa30665fbecfc1476d169ad1333d1

closed 2018-11-06 by aswaterman