riscv/riscv-isa-manual #222: 2.8, the semihosting comments

ilg-ul, 2018-08-18:
I appreciate that you added the semihosting workaround to the specs.

However, to be fully correct, I suggest you also explicitly mention that the 3 instructions are always the uncompressed version, even if the device supports compression.

You can do this in the text, but even more clear would be in the assembly code, by adding the `.option push/.option norvc` and `.option pop` assembler directives.

The actual code used for this looks like:

```c
  #define RISCV_SEMIHOSTING_CALL_NUMBER 7

  static inline int
  __attribute__ ((always_inline))
  os_semihosting_call_host (int reason, void* arg)
  {
    register int value asm ("a0") = reason;
    register void* ptr asm ("a1") = arg;

    asm volatile (

        // Workaround for RISC-V lack of multiple EBREAKs.
        " .option push \n"
        " .option norvc \n"
        " slli x0, x0, 0x1f \n"
        " ebreak \n"
        " srai x0, x0, %[swi] \n"
        " .option pop \n"

        : "=r" (value) /* Outputs */
        : "0" (value), "r" (ptr), [swi] "i" (RISCV_SEMIHOSTING_CALL_NUMBER) /* Inputs */
        : "memory" /* Clobbers */
    );

    return value;
  }
```

aswaterman, 2018-08-18:
Yes, good point. I’d favor writing it in English rather than including GAS directives in the spec.

ilg-ul, 2018-08-18:
Writing it in English is ok.

Thank you.

kasanovic, 2018-09-24:
The English text was added.

closed 2018-09-24 by kasanovic