riscv/riscv-isa-manual #226: SFENCE.VMA Before or After SATP Write

natestuder, 2018-09-13:
[This paragraph](https://github.com/riscv/riscv-isa-manual/blob/03a079586fcbcd37ae1728e716dfc3f9b633c996/src/supervisor.tex#L871-L875]) in the 1.10 Privileged Specification implies that a SFENCE.VMA instruction should be run before SATP writes to ensure TLB consistency. (emphasis mine)

> If the new address space’s page tables have been modified, it may be necessary to execute an SFENCE.VMA instruction **prior** to writing satp.

However, the [address space switching code for RISC-V in the Linux Kernel](https://github.com/torvalds/linux/blob/54eda9df17f3215b9ed16629ee71ea07413efdaf/arch/riscv/include/asm/mmu_context.h#L91-L92) does the opposite.

```
sr_write(sptbr, virt_to_pfn(next->pgd) | SATP_MODE);
local_flush_tlb_all();
```

So does the ordering of the SFENCE.VMA and SATP write matter as implied by the specification?

aswaterman, 2018-09-13:
Emphasis on "may be necessary."  The concern is, what if the mapping for the instruction immediately after SFENCE.VMA has been modified?   In the Linux kernel, this mapping is fixed (regardless of address space) so the concern does not apply.

The SFENCE after the SATP write is definitely necessary, though.  In general, you need to SFENCE after you've recycled an ASID.  Since we don't use ASIDs in the Linux kernel yet, every context switch is effectively an ASID reuse, hence the full TLB flush.

natestuder, 2018-09-13:
Thanks for the explanation.

closed 2018-09-13 by natestuder