riscv/riscv-isa-manual #227: Vector extension encoding conflicts

hawajkm, 2018-09-18:
I am working on implementing vector extension draft on a GNU toolchain. So far, I was able to successfully implement the complete draft; however, I had to improvise in couple of issues that I discover throughout the process.

The first issue was `op_code` encoding. The `op_code` chosen for most vector operation conflict with `JAL`. Although we can tell most instructions apart through bit 14, `VMADD`, `VMSUB`, `VNMADD`, and `VNMSUB` all have definite conflict with `JAL`. Also, `op_code` for all vector memory operations just mirror that of the float-point memory operations.

It seems that the opcode assignment were made as fillers? For now, I have used three of the reserved standard opcodes for vector-op, vector-load, and vector-store. I think that is not what the intent going forward; sacrificing three standard op-code extensions seems too much. But, for now, until the draft becomes more concrete, this would unblock my work.

The second issue is just a mere question: Would the immediate fields used in the draft be the same going forward? I am curious because the immediate field for load/store seems too small to encode anything useful and most likely the compiler will end-up filling it with zeros. Maybe it makes sense to make the immediate some multiple of a constant and/or it would be an upper-immediate?

aswaterman, 2018-09-18:
The encoding does not conflict, because MADD etc. require that m={1,2,3}. JALR corresponds to the case m=0. However, this encoding is temporary, and a revamped one will be released in the next month or so. Sorry for the confusion.

hawajkm, 2018-09-18:
I see. So, `m` can't be all zeros. Great!

However, does this mean that the loads/store are conflicting with floating-point loads and stores? I understand it is just temporary, I am just clarifying if I am reading the current encoding correctly.

Thanks!!

aswaterman, 2018-09-18:
The floating-point loads/stores have funct3 <= 4, and the vector ones have funct3 > 4 (since m > 0).

hawajkm, 2018-09-18:
Great! Thanks!

Is there any public mailing list/discussion forum that I can join to follow vector extension development?

aswaterman, 2018-09-18:
Much of the discussion happens in the Vector task group within the RISC-V Foundation.  You can join the Foundation as an individual member to participate (https://store.riscv.org/collections/membership/products/individual-member-1)

cbatten, 2018-09-18:
Hi Andrew, Khalid is a PhD student in my research group. What would be the best way for an academic to get involved in the Vector task group? Is an individual member the right choice?

aswaterman, 2018-09-18:
Hi Chris, good to know & nice to meet you, Khalid.

My understanding is that there are two equally valid options. Cornell, or
perhaps Cornell ECE or your research group, could join the Foundation as an
auditor, which costs $2500/year. This would allow anyone with that
affiliation to participate in the working groups where the sausage is made
(but does not connote voting rights, as do the pricier membership levels).
Some universities have chosen this option, and some have joined at the
voting levels of membership.

The other option is for individuals to join as individual members for
$99/year, with similar membership terms as the auditor level. The $99 can
be waived on a case-by-case basis for active open-source contributors.

cbatten, 2018-09-18:
Thanks Andrew! I think for now maybe Khalid will join as an individual member so we can be more in the loop for the vector spec and then we will go from there.