riscv/riscv-isa-manual #230: assembly code syntax for rounding mode

benjaminselfridge, 2018-09-21:
Is there assembly syntax for setting the rounding mode in a floating point instruction?

As a more general question, is there a definitive reference for RISC-V assembly? Is it the type of thing that's considered compiler-specific? Will the "Assembly Programmer's Handbook" contain the assembly code syntax for each instruction at some point?

aswaterman, 2018-09-21:
Specify it as a final argument, e.g. `fadd.s ft0, ft0, ft0, rtz`

The ASM manual is an incomplete work in progress: https://github.com/riscv/riscv-asm-manual/blob/master/riscv-asm.md

benjaminselfridge, 2018-09-21:
@aswaterman  How about the aq/rl bits for an AMO instruction?

aswaterman, 2018-09-21:
https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob;f=opcodes/riscv-opc.c;h=e0f711811f66a6e51abbe00dacf5d43cb1b27c33;hb=HEAD#l404

kasanovic, 2018-11-06:
Should be moved to be issues on the assembly language manual repo.

closed 2018-11-06 by kasanovic