riscv/riscv-isa-manual #233: misaligned instruction fetch exception ???

IC-Dream, 2018-09-28:
Hi,

For one core which only supports RV32I, all of the control transfer instructions(jump and branch)  would raise a misaligned instruction fetch exception if the target address is not aligned to four-byte boundry, right ? I can find the following description for jumps in spec, but the branchs doesn't document anywhere. 
[image]

Thanks
Dream

aswaterman, 2018-09-28:
In the section "Base Instruction Formats," it says "An instruction-address-misaligned exception is generated on a taken branch or unconditional jump if the target address is not four-byte aligned."

(Note that not-taken branches do not generate the exception.)

IC-Dream, 2018-09-28:
okay, Thanks.

closed 2018-09-28 by IC-Dream