riscv/riscv-isa-manual #239: mip view when an interrupt is delegated

rbohn, 2018-10-10:
I'm wanting to clarify what the intended behavior is for reads to the mip / sip csrs when one of the interrupt sources has been delegated to a lower privilege level. The text towards the beginning of the mip description states:

> **Restricted views of the mip and mie registers appear as the sip/sie, and uip/uie registers in
> S-mode and U-mode respectively**. If an interrupt is delegated to privilege mode x by setting a bit in
> the mideleg register, it becomes visible in the x ip register and is maskable using the x ie register.
> Otherwise, the corresponding bits in x ip and x ie appear to be hardwired to zero.

The text addresses what happens to the lower-privilege xIP registers when a given source is delegated, but may not address what happens to the higher-privilege xIP register. The statement "Restricted views of the mip and mie registers appear as the sip/sie, and uip/uie registers in S-mode and U-mode respectively" leads me to believe that sip / uip are just restricted views of mip and therefore if a bit is set in uip, for example, that same bit should also be set in mip.

However, later on in the PLIC chapter there is this text specific to the xEIP sources:

> Interrupt notifications generated by the PLIC appear in the meip/seip/ueip
> bits of the mip/sip/uip registers for M/S/U modes respectively. **The notifications only appear in
> lower-privilege xip registers if external interrupts have been delegated to the lower-privilege modes.**

Which I believe states that if, for example, UEIP were delegated from machine to user, then only the uip.UEIP would be set, while mip.UEIP would not be set.

To summarize, my question is: If an interrupt source has been delegated to a lower privilege level, does the higher-level xIP csr still indicate the presence of that interrupt to the higher-privilege level, or does the bit of any delegated interrupt always read 0 to that higher-privilege's xIP csr? And is the answer any different between the xEIP sources which appear to explicity state behavior in the PLIC chapter versus the timer / software interrupt sources.

If the answer is that the behavior mentioned in the PLIC chapter applies to xTIP xSIP as well, it might help clarify the specification to add the PLIC-chapter sentence "The notifications only appear in
lower-privilege xip registers if external interrupts have been delegated to the lower-privilege modes." to the 3.1.14 mip / mie text at the end of the paragraph I quoted.

aswaterman, 2018-10-11:
The delegated interrupt remains visible in the more-privileged xip register, and its enable remains toggleable through the more-privileged xie register.

If this weren’t the case, it would not be true that the less-privileged yip/yie is a “restricted view”.

rbohn, 2018-10-11:
Thanks for the explanation Andrew. Is it then the case that this text in the PLIC chapter is in error?

> Interrupt notifications generated by the PLIC appear in the meip/seip/ueip
> bits of the mip/sip/uip registers for M/S/U modes respectively. **The notifications only appear in
> lower-privilege xip registers if external interrupts have been delegated to the lower-privilege modes**.

You stated that the delegated interrupt remains visible in the more-privileged xIP register, yet the text in the PLIC chapter appears to explicitly state that it does not. (that's what I take "notifications only appear in lower-privilege xip registers" to mean).

aswaterman, 2018-10-11:
I see.  What the word "only" modifies in the sentence is ambiguous.  We meant "If external interrupts haven't been delegated, they don't appear in the lower-privilege xip registers."  You interpreted it as "If external interrupts have been delegated, they don't appear in the more-privileged xip registers."

I'll change the sentence to read, "The notifications don't appear in
lower-privilege xip registers if external interrupts haven't been delegated to the lower-privilege modes," which retains the intended meaning but removes the ambiguity.  Thanks for pointing this out.

closed 2018-10-11 by aswaterman

rbohn, 2018-10-11:
Thanks, I think that clarifies things. One last closing thought on this: in the slides that Allen presented in Barcelona there is a diagram (slide 42) that to me appears to imply that only a single pending bit is set. There are demuxes leading up to U / S / M "PEND" bits in the diagram. Perhaps the "PEND" bits in this diagram were not intended to be the literal xIP register bits, but instead a logical unexpopsed pending state that takes into account that "Delegated interrupts result in the interrupt being masked at the delegator privilege level".

commit 1038d230 (2018-10-11): Fix an ambiguity in PLIC spec

Resolves #239