riscv/riscv-isa-manual #240: sip / sie register description missing M*IP / M*IE bits

rbohn, 2018-10-11:
In another [issue](https://github.com/riscv/riscv-isa-manual/issues/153) a set of questions were raised about whether "machine" interrupts / exceptions can be delegated to lower privilege levels. In response to a question about whether MTIP becomes STIP when delegated to supervisor, Andrew stated:

> No, there's no crosstalk between these fields. **If MTIP were delegated, it would show up in the MTIP bit in sip.**

I think this points to a source of confusion in the current specification. Per the description in 4.1.5 "Supervisor Interrupt Registers" **there is no defined MTIP bit in sip.** The corresponding bit position is defined as WPRI, and the ensuing text goes on to describe the U*IP, S*IP.

Based on the responses I believe that sip / sie, as far as the specification is concerned, should have the same bits as mip / mie, with the visibility determined by mideleg as described in the mideleg description.

I think the idea that, for example, mideleg[MTIP] is a potentially-writable bit that allows delegation to a lower privilege mode, yet the lower-privilege xIP register defines the bits as WPRI, leads to a thought process that MTIP gets converted to STIP / UTIP when delegated.

rbohn, 2018-11-28:
> Some sort of clarification on this point is clearly necessary, as it's a
> frequent and understandable cause for confusion.
>
> Still not sure what the best fix is, though. I don't think the S-mode
> chapter is the right place to clarify it: adding MTI/MEI/MSI to sip/sie
> will be yet another source of confusion. I think we need to add a picture
> of how delegation works to the M-mode delegation chapter.

Maybe it's too late for this, but just changing the names of the bits I
think would go a long way towards minimizing sources of confusion. The M /
S / U prefixes create a lot of initial confusion when people are first
reading the spec I think. But they've been named like this for a long time
so I can understand not wanting to change.

I agree a picture would be good. Or even some pseudo code that shows the
relationship between *deleg and *ip / *ie registers (something that looks
like what an eventual formal model might show?).

aswaterman, 2018-11-28:
Indeed, changing the names isn't in the cards.  Maybe a picture with pseudocode in the caption.

commit ec2fe254 (2018-11-30): Add commentary about MTIP etc. in sip/sie

Closes #240
closed 2018-11-30 by aswaterman