riscv/riscv-isa-manual #243: Waiting for MIP.MTIP bit to be cleared after handling a timer interrupt

podhrmic, 2018-10-19:
The current privileged specs state that

> _The interrupt remains posted until it is cleared by writing the mtimecmp register._

But it is not specified how quickly **after** writing to `mtimecmp` the bit is cleared. In certain applications (such as FreeRTOS) `mtimecmp` is written to in the interrupt service routine (isr) and then the program immediately continues, assuming MIP.MTIP is cleared **before** exiting the isr. An example is [here](https://github.com/sifive/freedom-e-sdk/blob/FreeRTOS/software/FreeRTOSv9.0.0/Source/portable/GCC/RISCV/port.c#L240). 

However, in one of RISCV mailing lists it was concluded that a handler needs to wait until the MTIP clears. Having a busy wait loop in an interrupt handler in a real-time system is obviously undesirable, but  certain non-real time applications might prefer that to make sure the bit is cleared before continuing. 

I think this should be mentioned in the privileged specs as something to keep in mind. I am happy to make an appropriate PR if you agree that this should be explicitly mentioned.

aswaterman, 2018-10-20:
Just curious, if FreeRTOS gets a spurious interrupt because of this, what happens?

Note the spinning code is a little more complex than just looking at MTIP. What if MTIP never goes low because the new mtimecmp is so soon in the future that the vote never observes that mtime < mtimecmp? So you have to spin until either mtime >= mtimecmp or MTIP goes low. Undesirable, yes.

podhrmic, 2018-10-22:
I am assuming that the new `mtimecmp` is always set to some reasonable value, that is up to the user.
For example:

```
*timecmp = *mtime + (configTICK_CLOCK_HZ / configTICK_RATE_HZ);
```

For FreeRTOS, the default tick rate would be 1ms, but in general it can be anything from a few us to several ms.

aswaterman, 2018-11-21:
Thanks for raising this issue.  We've added a cautionary note in the spec: af46207cb79eda9d8eaf8237a38da68aec4232c4

closed 2018-11-21 by aswaterman