riscv/riscv-isa-manual #255: PMA implementation

dangerblue, 2018-11-07:
Hi,

about the PMA implementation, the spec did not specify how to implemente it in a SOC system. Normally the memory attribute are implemented in MMU, the cache can get the information such as non-cacheable, strong order information after access the MMU, also, the software can control the attribute in a fine granularity.

according to the spec, the PMA is not specified in MMU,  so, my question is how can the cache get the memory attribute information? does it allow the software to modify the attribute of a memory region?  if it is not allowed, then does it allow the software to read the attribute information?  and also, when one extendarable device such as PCIE，is not coherenced to the main memory, how to manage it?  because the device can be cachable or non-cacheable, but it cannot pre-define the attribute at the specified address.

if the PMA is a hardware implemented, such as the platform implement a memory attribute table, and the cache get the information though accessing the table region,  similar to access the PMP.  then the implementation cost is expensive.

please notice me if I have some mis-understanding about the PMA.    
thanks a lot

closed 2018-11-21 by aswaterman

rock-ifly, 2024-10-17:
1) Is the pma(Cacheable / shareable)  necessary? We think the answer is Yes
2) Are there any principles we should follow?

aswaterman, 2024-10-18:
Moving the discussion to #1687