riscv/riscv-isa-manual #263: Add Z extension description to chapter 28

jeceljr, 2018-11-09:
Chapter 28 describes "ISA Subset Naming Conventions". We now have three new extensions named Zicsr (chapter 9), Zam (chapter 23) and Ztso (chapter 24). I imagine the naming for this is similar to the X and SX extensions, but it would be good to explicitly define this in chapter 28.

closed 2018-11-26 by aswaterman

aswaterman, 2018-11-27:
I recall the same, but it might have just been a mailing-list proposal.  Certainly the `Zifencei` extension follows the pattern you've described, though ironically it's an exception to the rule, since FENCE.I is no longer proposed to be in I.

I agree we must standardize on a pattern for naming specific instructions within an extension, but I'd prefer to punt for the time being.  It's made somewhat more complicated by the desire to mnemonically name subsets of extensions (e.g. how do you name all the MUL* instructions without writing `Zmmul_Zmmulh_Zmmulw...`? you'd prefer to write `Zmm` or something).  Perhaps the rule is that subset names share the namespace of instructions within that extension.

asb, 2018-11-27:
Found the previous proposal: https://groups.google.com/a/groups.riscv.org/d/msgid/isa-dev/m2o9o4s31j.fsf%40dhcp-35-65.EECS.Berkeley.EDU

cc @kasanovic 

I'm totally fine with leaving this undefined for the time being, but obviously it's worth double-checking that the naming convention for the proposed Z extensions doesn't unduly limit us. There is a problem in that Ztso tramples over the namespace for a future 'T' extension. Maybe it should be Zztso?

commit 845b81f1 (2018-11-26): Add Z-extensions to naming chapter

Closes #263.