riscv/riscv-isa-manual #264: 'A' Extension LR-SC forward progress guarantee conditions may be onerous

wdc-pnl, 2018-11-09:
Implementation of guaranteed eventual success with the conditions specified in the second paragraph on page 45 may be excessively difficult.

Because there is no implicit increasing likelihood of success for each retry, eventual success requires some mechanism to force success of LR-SC (i.e. guarantee no further retries).

The 16 (64 bytes of) instructions are not required to be 64-byte aligned.  Requiring 64-byte alignment could help for those systems with 64-byte I-cache lines.  Systems with smaller cache lines, especially those with shared I-caches would have difficulty.

Data caching could have difficulty guaranteeing success (e.g. by holding onto write access for the accessed cache line for the duration of the LR-SC).

aswaterman, 2018-11-09:
Yes, it certainly can be onerous for some styles of implementation.  OTOH, we are of the opinion the instructions aren't useful without a forward-progress guarantee.

Requiring instruction alignment isn't really an option, because this part of the spec has been frozen for many years and software is not written accordingly.  Yes, it does imply that you eventually need to get hits across at least two I$ lines.  (The fact that RISC-V doesn't mandate I$ coherence makes this slightly easier, at least.)

All of the implementations I've worked on guarantee forward progress on the data side by holding off cache interventions for a short, bounded number of cycles.

aswaterman, 2018-11-30:
Not clear what is actionable here - LR/SC is hard to implement but removing the forward-progress guarantee makes it effectively useless.

closed 2018-11-30 by aswaterman

wdc-pnl, 2019-01-30:
I believe the pushback from implementors will become so strong that at some point we shall be forced to abandon this forward-progress guarantee.  The earlier this issue is addressed, the better it will be for RISC-V.  I don't believe that the lack of an eventual success guarantee makes LR-SC useless.   What other machine architecture offers such a guarantee?