riscv/riscv-isa-manual #266: Define 'G' ISA name as including Zifencei and Zicsr

asb, 2018-11-12:
I think it's expected that any "general-purpose" core provides these extensions, so it would seem consistent to maintain their presence in the RV32G/RV64G ISAs.

aswaterman, 2018-11-30:
Agreed.

closed 2018-12-20 by aswaterman

dkruckemyer-ventana, 2019-02-27:
There seems to be uncertainty in the memory model group (and possibly the J-extension group) discussions about the definition and implications of the FENCE.I instruction for certain classes of systems (e.g. large scale servers). Since these systems will most likely run Unix and the Unix platform working group will likely require RV64**G**C[...], this definition implies that all cores implemented for Unix will be required to implement FENCE.I as defined by Zifencei (as well as whatever else is defined to perform instruction cache maintenance).

Though this ship may have sailed, that requirement doesn't seem desirable....

dkruckemyer-ventana, 2019-02-27:
OK. My bad. I just reread the definition and commentary for Zifencei, and the definition of FENCE.I seems to be relaxed from what I remember from before (in particular, I thought there was a multiprocessor requirement, but that's not the case).

Never mind....

commit a7826272 (2018-12-20): Clean up naming and define G = IMAFDZicsr_Zifencei

Closes #266.