riscv/riscv-isa-manual #268: PLIC claim/response process

taoj33, 2018-11-15:
Hi,  I have a question about PLIC interrupt claim/response process.

The spec claims that
> It is always legal for a hart to perform a claim even if the EIP is not set. 

Does it mean the highest priority interruption can be claimed and handled even if masked by threshold? Is it more efficient that only the interruptions meet all requirements can be claimed?

And it may cause some problems based on my understanding. For instance, hart 0 and hart 1 both claim an interrupt notification. While claim from hart 1 is still in-flight, the claim from hart 0 may have already arrived and cleared its IP bit. When claim from hart 1 arrive, only the lower-priority interrupts are visible which may not satisfy the request to send out notification. If hart 1 gets response based on this info, it ends up responding to an interruption it intends to block. 

If there is solution to this issue or simply my misunderstanding, pls let me know. Thank you.

aswaterman, 2018-11-15:
To answer the other question @taoj33 asked - yes, it is possible to claim an interrupt whose priority is <= the threshold.  The only purpose of the threshold is to generate EIP.

taoj33, 2018-11-16:
@aswaterman @kasanovic  Thanks for ur reply. 
Still confused about following question: 
1. Do claims without EIP act same as ones with EIP, say clear gateway IP bit? 
2. Is it possible that the PLIC status can change after it send out its notification?  For example,after multi-casting a notification A, the PLIC samples a higher priority int source B, the claim response B instead of A ; or like the example I mentioned before, the int A is already claimed by other hart and only lower priority int C is visible, the claim response C, which may even lower than the threshold.

aswaterman, 2018-11-16:
EIP and claim are basically independent of each other. EIP means “something is pending, so you should try claiming it.” A claim after EIP goes high might return 0. It is inherently racy.