riscv/riscv-isa-manual #269: Clarify value in RD on misaligned jumps

olofk, 2018-11-15:
For a normal JAL(R), RD is updated with pc+xlen. Is this guaranteed also when we trap due to a misaligned effective address? Adding a line to clarify this would be good

kasanovic, 2018-11-15:
Misaligned address trap should be taken precisely, implying rd would not be updated.   This is same as for any architectural state update for a precise trap, so trying to clarify here would cause confusion everywhere else (or require every instruction description repeated what precise trap means).

asb, 2019-07-17:
Though the document doesn't explicitly say that the address misaligned exception is precise, does it? Misaligned loads and stores for instances are explicitly documented so they need not be precise (it documents the EEI may allow misaligned exceptions to be handled by a contained or fatal trap, which in turn may not be precise depending on the EEI).

A precise trap is the obviously sensible behaviour, but it's not clear to me it's actually enshrined in the current spec - unless I'm missing something?

aswaterman, 2019-07-17:
"The EEI defines for each trap whether it is handled precisely, though the recommendation is to maintain preciseness where possible." -- so it's up to the platform to mandate this.  I think that is the right decision.

asb, 2019-07-17:
Right, my understanding of the current spec is that this is clearly left up to the platform at least in the case of load/stores.

For this JAL(R) issue, Krste answered with a definitive statement that "Misaligned address trap should be taken precisely, implying rd would not be updated". If that's a requirement of the spec, I don't think it's clear in the current doc. If it's not, then the resolution of this issue is incorrect and it would probably make sense to update the relevant part of the spec to reference the relevant "trap" types introduced at the beginning of the doc.

aswaterman, 2019-07-17:
Yeah, I was implicitly agreeing with you :)  Krste implied a platform mandate that isn't substantiated by the unprivileged spec.  I still think it's appropriate to take no action in response to this issue, but not for the same reason that Krste closed the issue initially.  My reasoning is that, whether or not precise traps are provided, JAL/R is not special.

Platforms that permit imprecise traps definitely should circumscribe the permitted behavior, but IMO it is outside the scope of the unprivileged spec.

asb, 2019-07-17:
My thinking was that the description of JALR is special vs the one for misaligned loads/stores. For loads/stores, the spec goes to great care to talk about the interaction with EEI guarantees, discusses that it could be a contained or fatal trap etc. So wouldn't it make sense to be similarly detailed for JALR? How about documenting that it may produce a contained or fatal trap? [Which I _think_ is the intent, based on what you're saying]. That's better for the reader, as the beginning of the document already explains that these may or may not be precise, depending on the EEI.

aswaterman, 2019-07-17:
And the concern applies to JAL[R] because it both writes rd and can cause an exception?  (i.e., you don't view the spec for branches as being problematic?)

asb, 2019-07-17:
I hadn't really thought about the spec for branches - I came by this as we had some discussion about what the spec does/doesn't say about precise exceptions, and I found this issue searching "precise" in the issue tracker.

**Proposal**: as alternative to adding a large amount of prose to every other mention of exceptions perhaps to add a reference to EEI-dependent behaviour like (in the JAL/JALR case) "The JAL and JALR instructions will generate an instruction-address-misaligned exception if the target address is not aligned to a four-byte boundary, with any resulting trap behaviour dependent on the EEI."

This minimises excessive duplication, but gives a clear indicator to the reader that they should check another part of the doc for more details. It also gives a good indicator that it's incorrect to just assume precise exceptions, which I think i would tend to do if just reading the instruction definition part of the spec. It's easy to think "they must mean a precise exception, as that's surely what you'd want". That's true for most systems, but of course RISC-V targets a very wide range of design points and so the spec seems to not actually require this.

aswaterman, 2019-07-17:
The bit about "with any resulting trap behavior dependent on the EEI" part would need to be repeated every time the spec mentions an instruction raising an exception, or else we would create confusion if ever the phrase were omitted.  It strikes me as clutter.  I don't think it's too much to ask for people to read Chapter 1 first, or at least search the document for the word "exception" and read the section whose title contains that word.

I'm going to punt this one back to @kasanovic.