riscv/riscv-isa-manual #270: clarify which operand is signed vs unsigned for mulhsu

vbx-glemieux, 2018-11-17:
The MULHSU instructruction is used for "signed x unsigned multiplication" (section 6.1, p35 of the 2.2 draft up for ratification).

It's not clear whether the rs1 is signed and rs2 is unsigned (the natural order in which we write assembly language), or if rs1 is unsigned and rs2 is signed (the order the operands appear in the 32b encoding).

It is clear that rs2 is the "multiplier" and rs1 is the "multiplicand". This suggests that rd = rs2 * rs1, suggesting rs2 is the signed operand. However, this can also be written as:

   rs1  (multiplicand)
* rs2  (multiplier)
-----
   rd

which suggests rs2 is the unsigned operand, the opposite of what we concluded before.

So, which is it?

vbx-glemieux, 2018-11-17:
sorry, i was reading v2.2 as the latest pdf in github, assuming it was the same as the Nov 6 2018 draft being ratified. i guess that draft was only circulated by email and not in the releases folder on github.

the nov 6 draft is missing this MULHSU change in the changelog (if that's important).

aswaterman, 2018-11-17:
I'm going to claim this falls under the changelog entry "Improvements to the description and commentary."

Thanks for giving the spec a careful read prior to ratification, and please continue to do so.

closed 2018-11-17 by aswaterman