riscv/riscv-isa-manual #275: Move WFI encoding to volume I

ilg-ul, 2018-11-20:
WFI is a generic instruction. By defining its encoding in the Unix privileged specs, any other privileged specs will have to redefine it, leading to redundancy and possible inconsistencies.

Similarly to the ECALLand EBREAK encodings defined in volume I, WFI can also be defined in Volume I, with the definition striped from any mode-M, mstatus, etc specifics (the privileged behaviour can be explained in each privileged specs).

Profiles that do not need this feature can implement it as NOP.

And the name 'Interrupt-Management Instructions' is not very inspired. WFI is practically a power management instruction.

kasanovic, 2020-03-04:
We understand the argument that this could have been defined as an unprivileged instruction, with its use modulated by privileged CSRs.  For compatibility we'd define a new UWFI instruction that was unprivileged, but with the same effect, rather than change existing instruction, which has supervisor privilege.

jscheid-ventana, 2021-07-30:
Can this be resolved?

kasanovic, 2021-11-14:
We have a proposal for a WFMI, which as well as being an unprivileged instruction also is different in that it is allowed to continue sleeping while lower-privileged mode interrupt is pending-and-enabled.  This was developed in fast interrupts group and discussed here: https://github.com/riscv/riscv-fast-interrupt/issues/31

kersten1, 2024-07-03:
Assuming this has been dealt with and can be closed? If not, please reopen.

closed 2024-07-03 by kersten1