riscv/riscv-isa-manual #279: should vs. shall in floating-point rm field text

rbohn, 2018-11-26:
In the current base ISA spec there is this text:

Some instructions that have the {\em rm} field are nevertheless unaffected by the rounding mode; they **should** have their {\em rm} field set to RNE (000).

Would the word "shall" be a better fit than "should" in this instance? In other words, it is absolutely required that assemblers set the rm field to RNE for instructions that are unaffected the by the rounding mode?

aswaterman, 2018-11-26:
It is not absolutely required; these instructions execute the same way if `rm` is set to some other valid value.  So, "should" is the better wording in this case.

rbohn, 2018-11-26:
To help clarify can you give an example of an F extension instruction that has an rm field, is unaffected by rounding modes, "should" have it's rm field set to 000, but is legal to set the field to something other than 000? What specifically was this text going after?

Part of the confusion is that the spec is to me inconsistent with the field naming of bits 14:12. In the Chapter 25 table the four possible instruction formats all refer to these more generically as funct3, with only certain instructions calling those funct3 fields 'rm'. While the chapter 11 descriptions refer to all OP-FP and F[N]MADD/F[N]MSUB 14:12's as 'rm', with the fields being 'RM' for certain instructions. Some instructions use 14:12 as part of the decode and seem more appropriate to call the related field funct3 instead of rm (e.g. FMIN / FMAX). So from all this it's not clear which instructions "have an rm field" as far as this text is concerned.

I agree that instructions that are unaffected by rounding modes should obviously be unaffected, but the labeling differences between the chapter 25 table and the chapter 11 text make it unclear to me which instructions have rm fields and which just have funct3 fields that are part of the decode. 

One possible case I see where an instruction unambiguously has an rm field and does not round is FCVT.D.S. But this instruction is part of the D extension, not the F extension where the text I mentioned is located.

aswaterman, 2018-11-26:
"Have an rm field" = "Have a field named rm in the instruction listings."  So the comment doesn't apply to FMIN, for example.

The last paragraph you wrote is what we were going for.  Another example besides FCVT.D.S is FCVT.D.W[U].

aswaterman, 2018-11-26:
You're correct that no F instructions fit this category.

I'd like to leave the text where it is, but I agree it deserves some clarification.  Will push something shortly.

closed 2018-11-26 by aswaterman

rbohn, 2018-11-26:
Thanks Andrew.

commit 48994717 (2018-11-26): Clarify which instructions have RM field but don't use it

Closes #279