riscv/riscv-isa-manual #283: Re-evaluate the Z* extension naming scheme

asb, 2018-11-28:
Marking with the base ISA ratification tag because this should be evaluated prior to ratifying, as the opportunity to adjust the scheme is then lost.

See discussion in #263 for background.

A [previous proposal from Krste](https://groups.google.com/a/groups.riscv.org/d/msgid/isa-dev/m2o9o4s31j.fsf%40dhcp-35-65.EECS.Berkeley.EDU) suggested that a `Z[:alpha:]` would indicate a "category" for that extension, and could be used to name individual instructions.

Right now the manual defines the following Z extensions which don't all mesh with such a scheme. Obviously there's no real reason they should, but it's worth double-checking if the scheme meets our anticipated needs:
* Zifencei (fence.i is no longer an 'I' instruction though)
* Zicsr (added because CSR instructions are no longer in 'I')
* Zam (misaligned atomics, makes sense to have the Za prefix)
* Ztso (this would trample over the namespace for a future 'T' extension)

kasanovic, 2018-11-28:
My original idea had been that the first letter would give a category that all new instructions/subsets would be placed in, but categories are not the same as the mandated base for that category.  I did not intend that Zi, for example, would only be used as a prefix for instructions in the mandatory I base, but for all I-like instructions (hence Zifencei)
I realize there could be confusion about what is in I, but that can be captured with a Zibase or Zistd or some such name.

Requiring that every new instruction extension or individual  instruction (e.g, pause, fence.i) has a unique prefix, will require a longer string that might be harder to parse. e.g., when adding vector extensions or new floating point instructions (e.g., Zfsin).  Also, when discussing where an instruction lives during standards setting and ratification could change its string name, e.g., Zifencei to Z?fencei.

The Ztso was meant to live in the T category, as it is providing stronger memory semantics.  A base TM extension might still be named Ztm.

I think the canonical ordering would follow the category ordering to make it easy to locate instructions in a long string.

asb, 2018-11-29:
Categories with some degree of artistic license in determining which category an extension fits (e.g. "is it I-like") makes sense. To my mind Ztso seems to fit into that scheme oddly: TSO doesn't strike me as being "like" transactional memory. You could argue that CSRs are more S-like than they are I-like as well...

But anyway, it sounds like the chosen scheme should provide what is needed. I'm happy for this issue to be closed. Thanks for the response!

aswaterman, 2018-11-29:
Ztso's connection to TM is admittedly flimsy.  Zicsr is more defensible in that many unprivileged environments with I-only will still have access to performance counters.

Anyway, it seems like we're more or less in agreement.

closed 2018-11-29 by aswaterman