riscv/riscv-isa-manual #284: PLIC notification and claime issue

dangerblue, 2018-11-29:
Hi，
According to the spec，one interrupt notificates multiple core then each core attempt to claime the interrupt, and only one will claime successfully。Also, the threshold will not affect the claime operation, that is, the highest interrupt will be claimed even if its priority less than the coresponding threshold。

If I understand the spec correctly, I think there is an issue:

Assume two interrupts: int0, int1, with priority: int0 > int1
Two harts: hart0, hart1
The thresholds: th0, th1. And int0>th1>int1, int0>th0
Here is the process of the interrupt notification and claim process:
1. Int0 and int1 are both pending;
2. According to the threshold, the interrupt notification will send to both of the harts;
3. Hart0 and hart1 send claime operation to PLIC;
4. Hart0's claime operation arrive firstly, hart0 will claime successfully and get the information of int0;
5. The pending bit of int0 will be cleared, and the max id become int1;
6. Hart1's claime operation arrive. the claime will response to hart1 with int1. but, according to the thresold th1, the int1 cannot send the notification to hart1.
So, here is the problem：if there is no int0, the int1 will be masked by the threshold th1, However if there is int0, the hart1 will  bypass the threshold,

I think if the claime can be successful (get non zero value) after the threshold mask, not before, there will be no problem. 
also The functionality bellow can be  succefull by disable xie, not to set the threshold to the maximum value.

 `It is always legal for a hart to perform a claim even if the EIP is
not set.  In particular, a hart could set the threshold value to maximum
to disable interrupt notifications and instead poll for active
interrupts using periodic claim requests, though a simpler approach to
implement polling would be to clear the external interrupt enable in
the corresponding {\em x}{\tt ie} register for privilege mode {\em x}.`

thanks.

aswaterman, 2018-11-29:
Your description is right. This is correct behavior, by definition.

The threshold just guarantees that the interrupt won’t occur unless some interrupt exceeds the threshold. It doesn’t guarantee that the interrupt that is ultimately claimed will exceed the threshold.

closed 2018-11-29 by aswaterman

dangerblue, 2018-11-29:
Hi, the behavior here looks like the interrupt occur bypass the threshold. the processor will continue to handle the interrupt which is not supposed to break in.  It behaves not the way software expects. 
Is there a way to not make this happen? The behavior here is depending on the the absence of int0,

dangerblue, 2018-11-29:
OK, thanks