riscv/riscv-isa-manual #290: Can legal ISA naming strings omit extensions that are required/implied by another?

asb, 2018-11-30:
The D extension depends on the F extension. GCC will reject -march=rv32imd and Clang replicates this behaviour. Looking at Chapter 28: ISA subset naming conventions it doesn't actually indicate either way whether dependent extensions need to be specified. This is relevant for both standard and non-standard extensions.

See also: #289.

kiniry, 2018-11-30:
The feature model we have designed at Galois for RISC-V and its associated tooling—which we can demonstrate during the RISC-V Summit—supports the partial specification of configurations that semantically complete.  E.g., stating exactly that which you summarize above will imply that F is automatically included.

In previous work we have integrated such tooling into other tools and frameworks in order to leverage the feature model.  E.g., we have built command-line frameworks which have a built in feature model to enable completion of missing command-line options (exactly as you have highlighted), as well as bash command-line completion.   See, e.g., http://clops.sourceforge.net/janota2009clo.pdf and http://clops.sourceforge.net/

The place that readers may have incidentally seen this kind of work (by my colleagues) is in the Linux kernel configurator, which has an enormous built-in (automatically extracted) feature model of the kernel, as well as a SAT solver for supporting interacting configuration.  See https://www.kernel.org/doc/Documentation/kbuild/kconfig-language.txt for more information.

Early research work on this topic went so far as to denotationally specify the command-line switches of complex FLOSS tools like `gcc`, including the automatic generation of the necessary ASTs, parsing and pretty-printing code, etc.  You can think of that framework as an advanced version of `getopt`.  I'm excited to work with you, @asb, on integrating some of these ideas into the RISC-V clang backend in the coming months.

aswaterman, 2018-12-10:
I certainly see the motivation for @asb's question and Galois' approach, but OTOH I am a little concerned about baking these extension dependences into the standard tools.  At the moment, it is true that D implies F, and it seems very highly likely this will remain the case.  But what if, down the road, an important embedded application that only uses double-precision FP manifests?  If there's enough demand, the Foundation might want to specify that RV32ID (i.e. not-F) is valid.

asb, 2018-12-10:
@aswaterman: I don't really mind whether dependencies are implied or not, I just feel that in the interest of ensuring ISA strings are well-defined and interoperable between tools, the ISA manual should have something to say on this.

Options include:
* ISA strings with implicit dependencies are illegal
* ISA strings with implicit dependencies are valid, but discouraged as not all tools may support them
* ISA strings with implicit dependencies are valid and any tool consuming ISA naming strings should support them

aswaterman, 2018-12-10:
Understood.  In being mum on this matter, we had intended to imply the first option.  I'll just make this explicit in the spec.

commit c91cbde1 (2018-12-10): ISA strings can't leverage extension-dependence properties

Resolves #290
closed 2018-12-10 by aswaterman