riscv/riscv-isa-manual #296: Relation between I and E

kito-cheng, 2018-12-05:
`I` and `E` is different base extension, however `E` is just kind of reduced register version of `I`, but `E`  corresponding to which version of `I` seems like a problem.

The immediate problem is: `I` 2.1 no longer include `Zicsr` and `Zifencei`, so how about `E` 1.9? it's following `I` 2.0 or `I` 2.1?

aswaterman, 2018-12-10:
FENCE.I and CSRRx should not be part of the `E` base, which the text already implies by defining `E` in terms of `I`.  Since it doesn't mention a specific `I` version, it implies the latest version, which excludes those instructions.  Nevertheless, this is a change to `E`'s definition, so we probably should have incremented its version number...

When `E` becomes frozen, we should define a specific version of `E` in terms of a specific version of `I`, for example, "The only change between RV32I v2.2 and RV32E v2.0 is to reduce the number of integer registers."

Since `E` is not being ratified, I don't think it's urgent to address this, but I'll leave the issue open.

aswaterman, 2018-12-21:
Proposal: once E is ready for ratification, we will start numbering it consistently with I.

closed 2018-12-21 by aswaterman