riscv/riscv-isa-manual #297: LSB of observed PC

gameboo, 2018-12-05:
Hi.

When reading the JALR instruction definition, I see: 

>  The target address is obtained by adding the sign-extended 12-bit I-immediate to the register rs1, then setting the least-significant bit of the result to zero.

If I do a jump to some target address that would have had an LSB of 1 before the clearing of this bit, and land in some code that observes the PC (say AUIPC or a jump and link of some sort), it is unclear to me whether that observation should return always 0 in this LSB of PC or the value that I tried to jump to before the clearing of that bit. I presume that the intention is to preserve the bit rather than forcing it to 0 given what is written in the notes just below:

> Clearing the least-significant bit when calculating the JALR target address both simplifies the hardware slightly and allows the low bit of function pointers to be used to store auxiliary information.

but maybe it's not? In either case, it might be worth being explicit about it in the main text, or am I missing something?

gameboo, 2018-12-05:
Oh I see.

I guess that, *coming from the premise that I want to preserve the requested LSB in PC*, I ended up thinking of this *target address* (with zeroed LSB) as a value constructed solely for the purpose of performing the instruction memory access as opposed to being the value landing in the new PC.

Given that preserving this bit is not actually what is desired here, I can see why there is no need for this value to be different to the one used to update PC.

In that sense, I don't mean to suggest adding anything about sequential semantics, but rather something along the lines of replacing "the target address" with "the new PC". Although, as I am writing this, I realize the same terminology is used for the other control flow instructions. So maybe something at the beginning of this paragraph stating the obvious: "The target address of a jump or branch instruction is used to perform the instruction memory access *and* to update the PC register".

I also appreciate that having a 1 instead of a 0 in PC's LSB would otherwise always cause next IFetches to be unaligned if that bit is not deliberatly ignored when fetching instructions. So maybe this whole thing is not to be considered on the premise that we do care about PC's LSB when fetching instructions?

In any case, thanks for the fast answer.

aswaterman, 2018-12-05:
It’s not that we don’t care about the LSB of the PC - it’s that it’s always zero.

closed 2018-12-05 by aswaterman