riscv/riscv-isa-manual #298: Hypervisor guest time delta

aswaterman, 2018-12-06:
It has been requested that we add htimedelta[h] CSRs so that hosts can lie to guests about the current time, without requiring trapping and emulating.  There is no intent to add equivalent CSRs for cycle/instret.

https://groups.google.com/a/groups.riscv.org/forum/#!searchin/isa-dev/Time$20offset%7Csort:date/isa-dev/9F_-XcUd2kU/bhxHgb62BQAJ

cc @jhauser-us

bonzini, 2019-07-31:
Please do include the same CSRs for `cycle`, since the SBI set timer callback has absolute cycles as the argument. I agree that instret and performance counters can be handled via trap and emulate.

aswaterman, 2019-07-31:
sbi_set_timer is specified in terms of the time counter, not the cycle counter.  We're covered, as far as timer interrupts go.

commit 4bfae786 (2019-08-16): hypervisor: add performance counter delta registers

It has been requested that we add htimedelta[h] CSRs so that hosts can
lie to guests about the current time, without requiring trapping and
emulating. cycle is also included, since the SBI set timer callback has
absolute cycles as the argument.  There is no intent to add equivalent
CSRs for instret and performance counters.

Fixes: #298
closed 2019-08-16 by aswaterman