riscv/riscv-isa-manual #299: Clarification for exception flags for int to float conversions

kempfstefan, 2018-12-06:
Section 8.7 of the manual states that: "If the rounded result is not representable in the destination format, it is clipped to the nearest value and the invalid flag is set." Does this sentence refer to float-to-int conversions only, or this it cover int-to-float conversions as well?

My tests on x86 and with the Berkeley SoftFloat library raise an inexact flag when converting an int to a float if the int cannot be represented in the float exactly. The IEEE standard also states that an inexact flag is raised for the convertFromInt operation (section 5.4.1 of the standard).

Could you please clarify whether raising the invalid flag is expected for int-to-float conversion?

aswaterman, 2018-12-06:
It’s not about precision, it’s about range. Invalid is only raised if the number, after rounding, is too great in magnitude to be represented. If the result is merely imprecise, the inexact flag is raised.

aswaterman, 2018-12-10:
I looked at this text again, and

aswaterman, 2018-12-10:
I looked at this text again, and I think it is technically OK, because of the words "rounded result."  Since we've already rounded, the representability concern is not about exactness, it's about range.

I'm going to close this issue, but if you feel it's still problematic, feel free to re-open it and propose alternate text.

closed 2018-12-10 by aswaterman