riscv/riscv-isa-manual #300: Question regarding reserved category, HINT instructions and illegal exception

taoj33, 2018-12-08:
Hi,
In SPEC it defines Reserved as
 > Reserved encoding are currently not defined but are saved for future standard extensions

while for HINT, it is defined as following
> HINTS do not change any architecture visible state,except for advancing the pc and any applicable performance counters.

Is it correct to interpret that HINT is a subset of Reserved category which can be treated as NOP while  all other encodings in the Reserved category should be treated as illegal instructions?

aswaterman, 2018-12-10:
HINTs are not in the Reserved opcode category; they live within an already-defined standard encoding space.

You are right that HINTs should be treated as NOPs, whereas Reserved opcodes should raise illegal-instruction exceptions on current conforming implementations.

aswaterman, 2018-12-10:
-

taoj33, 2018-12-11:
@aswaterman  Thank you for your reply.

I can understand instructions with invalid func/opcode should trigger traps. 

But should instructions with undefined imm/reg raise illegal traps? for example, 
> C.JR  (RES, rs1=0)
> SLLIW,SRIW, and SRAIW encodings with imm!=0 are reserved

aswaterman, 2018-12-11:
Ah.  I understand the confusion, as the C chapter is written a bit differently and needs to be cleaned up.  The two examples you quoted are in different classes:

- The shift instructions with imm=0 are marked HINT in Tables 16.6 and 16.7, so they should execute as NOPs (which they do naturally, since 0 is the shift identity).
- C.JR with rs1=0 is marked RES, rather than HINT, so should raise an exception.

taoj33, 2018-12-11:
@aswaterman 
Sry made a typo 

> SLLIW, SRLIW, and SRAIW encodings with $imm[5] !=0 are reserved.

> Previously, SLLIW, SRLIW, and SRAIW with imm[5]=0 were defined to cause illegal instruction 
> exceptions, whereas now they are marked as reserved.  This is a backwards-compatible change.

So is it safe to conclude all instructions described as reserved should raise exception?

aswaterman, 2018-12-11:
Yes.  The difference between "is reserved" and "should raise an exception" is that, in the former case, you can put a non-compliant extension in that encoding space.  If you aren't using that space for a non-compliant extension, you should simply raise an exception.

asb, 2018-12-11:
Where in the spec does it say that instructions in reserved encoding spaces should trap? That's obviously a sensible and desirable thing to do, but I can't see the spec requiring it.

aswaterman, 2018-12-11:
Yeah, that’s why I said “should”; it’s not required. However, reserved encodings could eventually be used for new standard purposes. So implementations that don’t trap them could eventually become non-conforming to newer versions of the spec, since they are incapable of emulating the new instructions.

asb, 2018-12-12:
Thanks for the clarification, wasn't sure if you were using an IETF style SHOULD.