riscv/riscv-isa-manual #302: Fix RISC-V ISA naming everywhere

jhauser-us, 2018-12-11:
In the unprivileged document, section 28.3, "Instruction-Set Extension Names", it now says
> Some ISA extensions depend on the presence of other extensions, e.g., “D” requires “F” and “Q”
requires “D”. However, ISA strings may not omit the implied extensions, so, for example, RV32ID
may not be used to imply RV32IFD.

Extension F depends on Zicsr.  All instances of "RV32I..." designators that contain F need to be amended to include Zicsr, i.e., RV32...F...Zicsr".  For example, the sentence text above needs to say something like
> ...  so, for example, RV32ID may not be used to imply RV32IFDZicsr.

aswaterman, 2018-12-11:
Good point.  I've amended this new text accordingly, and I'll try to hunt down the other offending ISA strings.

aswaterman, 2018-12-13:
I've fixed some of them.  The rest are either benign, or depend on re-defining G to mean IMAFD_Zicsr_Zifencei, which is the subject of another pull request.

closed 2018-12-13 by aswaterman