riscv/riscv-isa-manual #305: ISA naming string ambiguity

aswaterman, 2018-12-13:
Because ISA names are case-insensitive, the versioning scheme creates an ambiguity with respect to the packed-SIMD extension.  Does the string `rv64ima2p2` mean I + M + Av2.2, or I + M + Av2 + Pv2?

kito-cheng, 2018-12-13:
Just a reference for you:

In current implementation of gnu assembler(Upstreamed), `p` will interpret as version separator if its following a number, otherwise interpret as `P` extension.

In order to prevent ambiguity, arch string will always separate with underline between different extension in current implementation of ELF attribute (Review in progress).

so for binutils, `rv64ima2p2` mean  I + M + Av2.2, `rv64ima2_p2` meaning I + M + Av2 + Pv2.

aswaterman, 2018-12-13:
Yeah, one easy solution is to require an underscore between a version number and the P extension.

commit 4096ba48 (2018-12-14): Disambiguate P extension in ISA strings

Closes #305
closed 2018-12-14 by aswaterman