riscv/riscv-isa-manual #307: Resolve D => F and F => Zicsr issue

aswaterman, 2018-12-14:
We want to maximize compatibility with previous versions of the ISA-naming scheme.

Proposal for Zicsr is that F implies Zicsr, so RV32IF is equivalent to RV32IFZicsr.

Re: D => F, JohnH suggested:

Make a note in the document that, although the D extension is not
currently defined without F, it is intended that a future version of
the specification will define D separate from F (but such that the
combination F + D remains unchanged).  Hence, "D implies F" is only
temporary.  As such, a special case applies that you must write an ISA
string as though D did not imply F, because in the future it won't.

If necessary, change "will define" to "may define", etc.

asb, 2018-12-14:
When dependencies are explicit in the ISA naming string we need to be careful of introducing new dependencies in future spec revisions. When they are implicit, we need to be careful of removing new dependencies in future spec revisions. Unless it's obvious whether adding/removing dependencies is likely more obvious, it seems it doesn't matter too much which way we go.

One additional point for consideration: implicit dependencies require more smarts when processing an ISA naming string. My software tool might have no idea what Zicsr is, but if I ask it whether RV32IFZicsr implements Zicsr then it can tell me. Not necessarily the case if it has to check its internal dependency info, which might not be up to date.

aswaterman, 2018-12-14:
My current thinking is as follows:

- The name "RV32IF" should continue to work right.  If tools start requiring RV32IFZicsr, that will pose a significant usability problem.  Rather than make this a special case in the tools, I think it should be formally accepted that F implies Zicsr.

- I don't think D-sans-F is a serious likelihood.  It was just an example to help us hammer out this problem.  So, I'm reluctant to remark that "D implies F" is a temporary thing.

- I'd rather have only one type of dependence (explicit or implicit, but not both).  Since F -> Zicsr is implicit, that suggests that D -> F should be, too.  Current tools don't understand this, but at least when they are brought up to date, they will continue to accept the explicit strings.

What do you think?  I'd really like to stop thinking about this issue today.

aswaterman, 2018-12-14:
Here's the change I'm proposing: https://github.com/riscv/riscv-isa-manual/commit/619b1938bdb2da7fd97853d61e767f455cf2a62a

closed 2018-12-20 by aswaterman