riscv/riscv-isa-manual #324: Hypervisor CSR addresses

avpatel, 2019-01-08:
Please provide exact 12bit addresses of hypervisor CSRs. Currently, this is blocking us to add hypervisor extension in emulators/simulators (i.e. QEMU).

aswaterman, 2019-01-08:
Here is the tentative assignment (note it’s not frozen):

   200  bsstatus
   204  bsie
   205  bstvec
   210  reserved for bsstatush
   214  tentatively reserved for bsieh
   240  bsscratch
   241  bsepc
   242  bscause
   243  bstval
   244  bsip
   254  tentatively reserved for bsiph
   280  bsatp
   A00  hstatus
   A02  hedeleg
   A03  hideleg
   A12  tentatively reserved for hedelegh
   A13  tentatively reserved for hidelegh
   A80  hgatp

avpatel, 2019-01-09:
Thanks, we will use tentative CSR addresses for now.

BTW, what are some of these CSRs with "h" suffix namely bsstatush, bsieh, bsiph, hedelegh, and hidelegh ?

aswaterman, 2019-01-09:
Those hold the upper 32 bits of a CSR for RV32 only. They aren’t fully spec’d yet, and you can ignore them for RV64.

aswaterman, 2019-01-22:
Closed by https://github.com/riscv/riscv-isa-manual/commit/516b9ed10b7c3bf17593034b66bd2c203d26a510

closed 2019-01-22 by aswaterman