riscv/riscv-isa-manual #332: THE xEPC[1] "HIDDEN STATE" PROBLEM:

rsnikhil, 2019-01-28:
[ Apologies that this is submitted after the official ratification period.]
[ This is collectively from the ISA Formal Spec Technical Group. ]

Hidden state complicates formalization.

(And therefore not surprisingly also complicates virtualization's save/restore).

In the proposed spec, xEPC[1] can be hidden state (bit [1] of MEPC/SEPC/UEPC).

According to the spec, Sec. 3.1.20 on MEPC (similarly Sec. 4.1.9 on SEPC):

   "... whenever IALIGN=32, bit mepc[1] is masked on reads so that it
    appears to be 0. This masking occurs also for the implicit read by
    the MRET instruction. Though masked, mepc[1] remains writable when
    IALIGN=32."

Consider a sequence like this:
1. A CSR instruction writes 0 to MISA.C
2. CSR instructions change MEPC [1] by writing to MEPC

While MISA.C is 0, we can no longer read the effects of (2) on MEPC[1],
since it is always masked to 0, i.e., it's become a piece of hidden state.
    
Of course, the bit can be made visible by setting MISA.C again, but
this is messy (amongst other things, one has to be in M mode for this).

>----------------
PROPOSED FIX:

In Sec 3.1.2, add a new last paragraph:

    "An attempt to disable the "C" extension (by writing 0 to MISA.C)
     is ignored if MEPC[1] is currently 1, i.e., MISA.C is left unchanged."

This would be similar in spirit to the existing last paragraph:

   "Writing misa may increase IALIGN, e.g., by disabling the “C”
    extension. If an instruction that would write misa increases
    IALIGN, and the subsequent instruction’s address is not IALIGN-bit
    aligned, the write to misa is suppressed, leaving misa unchanged."
>----------------

aswaterman, 2019-01-28:
I agree the hidden state is unfortunate, but I don't think we can get away with making this backwards-incompatible change.  Furthermore, it took a lot of wrangling to get people to settle down on the current definitions, and I'm not personally willing to re-open those wounds.

jhauser-us, 2019-02-22:
The `*epc[1]` bits might not be the only state that can be selectively hidden and then later revealed through changes to `misa`.  Consider, for example, a processor that selectively supports S mode, so `misa.S` is writable.  When `misa.S` = 1, register `mstatus` must have visible writable bits TSR, TVM, MXR, SUM, SPP, SPIE, and SIE, but when `misa.S` = 0, those bits are required to read as zeros.  What happens with this sequence?:

- Write `misa` so bit S = 1.
- Write `mstatus` so bits TSR, MXR, SPP, and SIE = 1.
- Write `misa` so bit S = 0.
- Write `mstatus` with a value having bit positions TSR, MXR, SPP, and SIE = 0, and bit positions TVM, SUM, and SPIE = 1.
- Write `misa` so bit S = 1.

If you next read `mstatus`, what should you see in bits TSR, TVM, MXR, SUM, SPP, SPIE, and SIE?  The document doesn't really say.  You may feel these fields should obviously all read as zeros at this point.  However, I know of some influential folk who are more concerned about saving a few gates and want it to be acceptable (maybe even required) that the value read from `mstatus` has either TSR, MXR, SPP, and SIE = 1 or TVM, SUM, and SPIE = 1.  (Which is it?  I suppose the latter, but you'd have to ask them.)

There are more examples of the same effect, not clearly resolved in the ISA documents.

Given the few number of RISC-V cores implemented so far having writable `misa` (I know of none, but I'm not aware of everything going on), it's probably not too late to settle this definitively one way or another.  But you'll have to act quickly, and find some more allies to join you.

When I participated in the discussion about the relationship between `misa.C` and the `*epc[1]` bits, I argued that successfully setting `misa.C` to zero should also force the `*epc[1]` bits to zeros, to avoid the hidden state.  But I was overruled by concerns about saving a couple of gates.  I could accept a weaker spec that says `*epc[1]` becomes unspecified when `misa.C` is changed from 0 to 1.  However, the current *required* behavior seemed like a crazy choice to me, since the few gates saved are totally insignificant.  But, like Andrew, at the time I wasn't willing to upset the delicate apple cart over the issue.  It may be possible now to go back and relitigate this particular detail again, if there is enough complaint.

marceg, 2019-02-24:
It is entirely appropriate to leave certain results undefined (and preferable to explicitly mention such undefined cases for clarity, rather than say nothing about them).  In other words, leave the implementation to choose the result, and make it clear that validation tests can ignore the result.  Why not do this here, leave EPC[1] undefined when MISA.C re-enables it?  Was a perceived need for completely deterministic results across all implementations any part of that wrangling?

aswaterman, 2019-02-24:
@marceg As a design principle, we’ve tried to avoid undefined behavior unless there’s a compelling reason to the contrary.  In this case, requiring that everyone implement this feature the same way didn’t seem too heavy-handed. Although there was significant disagreement over what the behavior should be, there is very little hardware cost or design complexity implied by the chosen specification.

jhauser-us, 2019-02-25:
The differences in hardware between these different choices are essentially in the noise, I think.  So the question should really be what's best for software and for testing.  Is it better to require hidden state that hardware must maintain a specific way even though it's not visible, or is it be better (usually) to say that when register bits are newly exposed that they take unspecified values?

I have an opinion (unspecified is usually better), but I'm willing to defer to our experts on verification and compliance testing if we have some consensus from that front.  Do we?  Have we asked?

The issue author, rsnikhil, also expressed concerns for virtualization that I'm not sure should be dismissed.  When swapping contexts, hidden bits can't be properly saved and restored without software knowing that they're there (or might be there), and knowing how to make them visible temporarily if they are.  But if the values of newly exposed bits are defined as unspecified, hidden bits won't normally need to be saved and restored on context swaps.

ben-marshall, 2019-03-13:
>  In other words, leave the implementation to choose the result, and make it clear that validation tests can ignore the result.

Implementation defined cannot mean that a test just ignores the result. For co-simulation based verification, you would need to extract that value and hint it too your golden model. For formal methods, you end up needing hooks in the model so you can alter it to an individual implementation of the spec, which is not desirable if the formal model is supposed to be *the* reference/spec.

> I have an opinion (unspecified is usually better), 

The words "unspecified", "undefined" and "implementation specific" always make verification and compliance much harder. Every bit of freedom you give a hardware designer is another possible result/behaviour you must check for as a verification engineer.

jhauser-us, 2019-03-13:
Ben Marshall wrote:
> The words "unspecified", "undefined" and "implementation specific" always make verification and compliance much harder. Every bit of freedom you give a hardware designer is another possible result/behaviour you must check for as a verification engineer.

Nevertheless, as it currently stands, the behavior around hidden state is de-facto *unspecified*, because the document doesn't specify what is required for cases like the example I gave above involving `misa` and `mstatus`.  The longer the document remains silent on such matters, the more likely we will have actual RISC-V implementations that do different things, at which point retroactively adding a requirement will render some implementations noncompliant, which they may fight against.

I'll ask again:  Have our experts on verification and compliance testing had a discussion about architected hidden state and reached a consensus opinion that can be acted on?  If people with the relevant clout don't act on this, *unspecified* is what we're all going to get, whether you like it or not.