riscv/riscv-isa-manual #333: A PROBLEM WITH THE DEFINITION OF 'WARL'

rsnikhil, 2019-01-28:
>----------------
A PROBLEM WITH THE DEFINITION OF 'WARL':

[ Apologies that this is submitted after the official ratification period.]
[ Also, I seem unable to add the "Priv Arch Ratification" label to this issue on GitHub.]
[ This is collectively from the ISA Formal Spec Technical Group. ]

The definition of WARL is too restrictive.

According to the spec, Sec. 2.3 on CSR Field Specifications:

   "Implementations can return any legal value on the read of a WARL
    field when the last write was of an illegal value, but the legal
    value returned should deterministically depend on the illegal
    written value and the value of the field prior to the write."

The problem is in the last phrase, 'deterministically depend on ...'

The definition of legality may depend on dynamic state other than the
attempted write-value and old value.

Examples:
- Exception codes in MCAUSE:
    - AMO exception codes legal only when MISA.A is 1
    - Page fault exception codes legal only when MISA.S is 1
    - ECALL-from-S exception code legal only when MISA.S is 1

- MODE field in SATP:
    - Sv32/Sv39 legal only when MISA.S is 1

- Clearning MISA.C is ignored when not 32-bit aligned (cf. last para
    Sec 3.1.1) i.e., depends on current value of PC.

>----------------
PROPOSED FIX:

The spec should specify, for each WARL field, exactly which other
state is relevant and can be examined to determine the legal value
returned subsequently.
>----------------

aswaterman, 2019-01-28:
Should we revise the definition of WARL to say that the value should "deterministically depend on the illegal value, the value of the field prior to the write, and any other architectural state explicitly mentioned in the definition of the field?"

By the way, the mcause exception-code field is WLRL, not WARL, so happily those comments do not apply.

aswaterman, 2019-02-22:
Note to self: the misa bits U and S can affect the WARLness of mstatus.MPP.

neelgala, 2019-02-23:
Agree completely with the issue raised by Nikhil.

I just wanted to point out that MISA in itself is optional - an implementation can support rv64imafdcnsu and have misa hardwired to 0. In such a case, the legal-next-value of mstatus.MPP will not depend on MISA but rather could depend on "platform defined non-standard csr/state" of the machine. 

Hope this aspect is considered while redefining some of these fields in the spec.

Would it be really bad to now switch MISA to a "mandatory" CSR instead of "optional" ? I know the compliance group would be happy with this change :) 

@rsnikhil 
btw, if you have MISA.A set to 0, you would never get an amo-exception since all amo instructions would be treated as illegal. isn't it?

aswaterman, 2019-02-23:
Sorry, not gonna happen. Existing implementations don’t implement it.

aswaterman, 2019-02-23:
I should expound. misa is optional because requiring it slightly increases the cost of a minimal implementation, but it doesn’t provide any value in the case that software is written knowing what ISA extensions are present.

You’re of course right to point out that MPP only depends upon misa if misa is implemented, and these clarifications to the spec should reflect that fact.

kasanovic, 2019-02-24:
Making misa mandatory can be part of a platform profile, e.g., a security monitor profile could require that it exists to check for certain instructions being present to simplify verification of software code.

neelgala, 2019-02-24:
I agree with your points.
My concern was simply from the aspect of writing compliance tests for the privilege spec. If misa is optional, the tests now have to probe multiple other states/csrs to even know if a particular extension/feature is available/supported on the platform. This might tend to either making the tests complex or maintaining multiple tests for a feature - each with a different assumption.

But nonetheless, it would be good to update the spec with dependencies of each WARL field for better understanding.

allenjbaum, 2020-01-03:
The riscof compliance framework has defined a WARL syntax which seems to cover all the cases I can think of, including fields whose definition changes depending on other fields, e.g. xTVEC can be limited to page granularity or not, depending on the interrupt vector mode.
see: 
      https://riscv-config.readthedocs.io/en/latest/yaml-specs.html#warl-field-restriction-proposal
I propose we start with this; I believe it should cover all WARL definitions that anyone has actually implemented, including some odd ones, like TVAL in spike which is the illegal address with a flipped MSB.

aswaterman, 2020-01-03:
As discussed over email, I don’t think listing out all of the dependences that set which values are legal is tractable in the written spec without reducing clarity due to bloat. (Think of all the fields whose legal values depend on misa.S or misa.H.) This is the provenance of the formal spec.

We should, of course, correct the definition of WARL to indicate dependence on other state, perhaps to say the new value “deterministically depends on the illegal value and the architectural state of the hart”.

allenjbaum, 2020-01-03:
Perhaps we could simply differentiate WARL with no dependence on other
state, and WARLd with dependence?
Then, at least, you'd know to look elsewhere.

allenjbaum, 2020-01-03:
Oh, and I agree about the bloat listing all dependencies would cause; that
should be discernable formal spec.
Ideally, the written spec would have live links to the formal spec code
that defines the behavior of the CSR.

closed 2020-03-04 by kasanovic

VizorKit, 2021-01-05:
Could someone link me to the full definition? I didn't even deduce what WARL stood for in this discussion. The best I could gather is from fe310-g002 chapter 10.6 'threshold' is a warl field, and it sounds like from this discussion that nothing can be illegally written ever as the threshold is 3 bits wide and all possible states are legal, so how can it be warl if there is no possibility of there being an illegally written value?

allenjbaum, 2021-01-05:
Legal values are implementation specific. The meaning of those values are architectural.
So, in the case of your 3bit fields with 8 meanings, your implementation may only allow a 2 values (with architecturally defined semantics). Trying to write any of the other values will result in one of those 2 values being written. 
The mapping from illegal->legal is also implementation defined and can be any deterministic function of the current state of the Hart and the value being written.

The Architectural Test TG (formerly Compliance TG ) has defined a syntax for those mappings and any mapping that doesn’t match will not pass our tests (yes, despite the fact they are allowed by the spec).
We recommend that the mapping that leaves the existing legal value unchanged be the one that is used (and is probably the easiest to implement)

-Allen

commit fc44561b (2020-03-04): Refined definition of WARL.
Closes #333.