riscv/riscv-isa-manual #347: Clarifying PMP and superpages

lukenels, 2019-03-02:
My observation when running code that uses PMP on the HiFive Unleashed board is that, if an access results in a page fault that resolves to a superpage, PMP checks are performed for the entire extent of the superpage. For example, an access inside a 1GB page will check that the entire 1GB range is accessible under the current PMP configuration. 

As a concrete example, we constructed two PMP regions that allow access to supervisor-mode code and page tables. The page table is set up to provide an identity mapping, using 1GB superpages. Our observation is that this causes a PMP access fault, we suspect the hardware is checking the entire 1GB range on TLB miss. This behavior does not occur on Spike, and it is unclear whether this behavior is permitted by the specification or not.

The privileged specification mentions something close to this case in section 3.6.2, however this only mentions the checks on implicit references to page tables and not to the entire leaf page itself:

> When paging is enabled, instructions that access virtual memory may result in multiple physical-memory accesses, including implicit references to the page tables.

If this behavior is allowed, we suggest that the specification clarify that PMP checks may be more strict than expected when paging is enabled. In particular, PMP accesses may be performed on entire pages on TLB miss rather than just on the access itself.

aswaterman, 2019-03-02:
Mind sharing the code (particularly in a way that I can readily run it on Spike)?  I'd like to investigate the discrepancy.

lukenels, 2019-03-02:
Thanks for the quick reply! It will be hard to carve out this test from the rest of our testing infrastructure as its part of a large project, but here is the code that performs the basic setup operations that demonstrate the discrepancy between Spike and the HiFive Unleashed (this passes on Spike but fails on HiFive).

I'm happy to give any more information you need or to send you a binary over email if that would help.

```c
static int exec_superpage(struct test *test)
{
        struct pt_regs regs = { 0 };
        unsigned long cfg0, addr0, addr1;
        static pgd_t pgd[512] __aligned(SZ_4K);
        uintptr_t addr = (uintptr_t)test_empty;
        size_t i;

        /* delegation off */
        csr_write(medeleg, 0);
        csr_write(mideleg, 0);
        /* 1G identity mapping */
        for (i = 0; i < ARRAY_SIZE(pgd); ++i)
                pgd[i] = pfn_pgd(PHYS_PFN(i * SZ_1G), PAGE_KERNEL_EXEC);
        /* paging on */
        csr_write(satp, SATP_MODE_SV39 | PHYS_PFN((uintptr_t)pgd));

        csr_write(mstatus, SR_MPP_S);

        cfg0 = csr_read(pmpcfg0);
        addr0 = pmpaddr_read(pmpaddr0);
        addr1 = pmpaddr_read(pmpaddr1);

        /* clear PMP configurations */
        csr_write(pmpcfg0, 0);
        pmpcfg_write(pmp0cfg, PMPCFG_A_NAPOT | PMPCFG_X);
        pmpcfg_write(pmp1cfg, PMPCFG_A_NAPOT | PMPCFG_R | PMPCFG_W);
        /* 4K code region */
        pmpaddr_write(pmpaddr0, (addr | (SZ_4K / 2 - 1)));
        /* 4K pgd */
        pmpaddr_write(pmpaddr1, ((uintptr_t)pgd | (SZ_4K / 2 - 1)));
        local_flush_tlb_all();
        test_run_guest(&regs, test_empty);

        csr_write(satp, SATP_MODE_BARE);
        csr_write(pmpcfg0, cfg0);
        pmpaddr_write(pmpaddr0, addr0);
        pmpaddr_write(pmpaddr1, addr1);

        ASSERT_EQ(EXC_ECALL_S, csr_read(mcause));

        return 0;
}
```

aswaterman, 2019-03-02:
Thanks - I'll send you an email with the gory details if I can figure them out, and we can keep this issue open for the purposes of clarifying the spec if need be.

aswaterman, 2019-03-02:
As to the matter of the specification, since we write that PMP is designed to compose with VM, and we don't explicitly allow this re-interpretation of the PMPs, I'd say that the behavior you describe (PMPs must be homogeneous across a superpage, or else the access fails) is implicitly invalid.

aswaterman, 2019-03-02:
I think this is resolved, but feel free to reopen if there's more to discuss

closed 2019-03-02 by aswaterman