riscv/riscv-isa-manual #350: Recursive virtualization in S-mode does not work well with global page.

nbdd0121, 2019-03-04:
In current design, a S-mode recursive supervisor will suffer in performance even if it utilises ASID, due to global pages. In order to ensure correctness, it needs to either
* shadow all global pages as non-global - which is clearly undesirable because the TLB will now creates multiple copies of these global pages in TLB, and all SFENCE.VMA will flush all global pages.
* flush all global pages when switching from a guest to another guest - which is also undesirable because it will also have huge performance impact.

aswaterman, 2019-03-04:
The overhead of maintaining the shadow page tables will usually dominate, and so this will not be a major issue in most use cases.  The first option is acceptable.

This part of the spec is frozen.

closed 2019-03-04 by aswaterman

marceg, 2019-05-12:
I can't speak for the performance impact spoken above.  If it turns out to be significant, here's a quick napkin sketch of an idea ...
The Xtensa architecture MMU happens to solve this by allocating a separate ASID for global pages.  The TLBs match against two possible ASIDs (actually, up to four for Xtensa, although two is generally plenty).  It may be relatively trivial to specify a new optional global-ASID register, independent from the existing non-global ASID.  If left untouched, there is no impact on software, other than M-mode needing to context-switch it among multiple supervisors.  It can be written, thus switching to a different group of global pages.  Writing it atomically with SATP requires some solution, such as two such global ASIDs (the currently active one and the one to take effect upon the next SATP write), or a new instruction that writes both SATP and the global-ASID.  The former of these two seems more backwards compatible.  Also to maintain compatibility, global ASID space can be kept separate from non-global ASID space.  If TLBs already have a global bit, this does not increase the amount of TLB state, it makes more efficient use of that state.  The TLB global bit becomes a selector of which ASID to match with.  It's also not clear offhand whether you'd want this global-ASID writable only by M-mode or also by S-mode.  A standard numbering for global ASIDs unique from normal ASIDs might be established so that SFENCE.VMA can specify a global ASID.