riscv/riscv-isa-manual #353: Unsigned Int in RV64

kasanovic, 2019-03-07:
Starting new issue with first part of question from Alibaba.
"1. Normally, RV64 treats 32-bit data as signed data. This means that when 32 bits are involved in the arithmetic, MSBs are always sign extended to 64 bits. However, this is a disaster for programs such as coremark that use unsigned int as an array index. The array index needs to be zero extended to 64 bits by slli and srli, and then the base address is added to execute the load operation.
A well-known trick is to change the definition of unsigned int to signed int, which is not explicitly prohibited in earlier versions of coremark. With this trick, the coremark performance of the high-performance RV64 processor can be improved by more than 14%.
However, last month EEMBC updated the coremark requirements to make sure that unsigned data type must be treated as an unsigned number:
eembc/coremark@7663f78
That is, EEMBC believes that 64-bit architecture should provide adequate support for unsigned int types. This is a realistic need in real world.
By adding indexed load and unsigned offset, you can compensate for 14% of coremark performance loss. Thanks to indexed load and store, the total performance improvement can reach more than 20%.
Does RV64's base ISA have a plan to solve this problem?"

kasanovic, 2019-03-07:
The base ISA itself (RV64I ) won't be changed to support this, as this has been frozen for a long time and a change would be disruptive.  Unsigned 32-bit arithmetic support could be proposed as part of an ISA extension.  For systems supporting only base ISA, one approach is to work on compiler analysis to determine where the zero-extension can be elided for array indices.

For the specific Coremark case, using an unsigned int as an array index is poor C coding practice, as size_t is explicitly defined in C standard to be used for this purpose in portable code.   While I agree with argument that a system should run poorly written code well, I've seen int used as array index far more often than unsigned int.  While Coremark is obviously important commercially as an embedded performance benchmark for marketing, I don't believe this particular use of unsigned 32-bit value is common in real-world code. 

I don't want to speculate on why EEMBC chose to enforce the unsigned rule as opposed to fixing the source code to follow C standards, or changing the variable to signed int to provide a more representative example of poor coding style.

nbdd0121, 2019-03-10:
>  as size_t is explicitly defined in C standard to be used for this purpose in portable code

Many code actually use `int` for array indices not `size_t`, so I speculate that there might be some cases where `unsigned int` variable is also occasionally used for indexing. But that should be pretty rare.

I think for non-byte arrays this isn't a big issue - we only change the sequence from 'slli' to 'slli, srli', so only 1 extra instruction. For compiler analysis part, I did a quick check with GCC, and it looks like that it cannot yet use range information to elide zero extension.

```c
char test(char* num, unsigned v) {
    if (v>1000)return 0;
    return num[v];
}
```
will produce
```
test(char*, unsigned int):
  li a5,1000
  bgtu a1,a5,.L3
  slli a1,a1,32
  srli a1,a1,32
  add a1,a0,a1
  lbu a0,0(a1)
  ret
.L3:
  li a0,0
  ret
```
where the two shifts can actually be elided.

jnk0le, 2019-03-10:
Register offset store would be the only instruction requiring 3 read ports.

Presumably something like ARMs uxta* (+explicit offset scaling) in B extension could replace that slli,srli,add sequence with a single instruction.

dennisyyz, 2019-03-11:
> The base ISA itself (RV64I ) won't be changed to support this, as this has been frozen for a long time and a change would be disruptive.  Unsigned 32-bit arithmetic support could be proposed as part [...]

Do you mean by eliding zero extension based on the array index boundary check？That's a possible solution but currently no similar optimizations are seen in the compiler.

> Register offset store would be the only instruction requiring 3 read ports.
> Presumably something like ARMs uxta* (+explicit offset scaling) in B extension could replace that slli,srli,add sequence with a single instruction.

For those who don't want indexed load/store, adding addition of zero extended scaled offset is an effective attempt . We have tried that and saved about 10% of the performance loss.

jim-wilson, 2019-03-14:
On Mon, Mar 11, 2019 at 4:12 AM dennisyyz <notifications@github.com> wrote:

> The base ISA itself (RV64I ) won't be changed to support this, as this has
> been frozen for a long time and a change would be disruptive. Unsigned
> 32-bit arithmetic support could be proposed as part [...]

Looking at our competition, I see that power64 and sparc64 store unsigned
32-bit values zero-extended in a 64-bit reg, and signed 32-bit values
sign-extended in a 64-bit reg.  So neither requires an extra zero extend
here after the compare.  These targets both have 32-bit and 64-bit compare
instructions though and RISC-V has only 64-bit compares.   Aarch64 has an
addressing mode that can add a sign/zero extended 32-bit value to a 64-bit
value, so no extra instruction here.  x86_64 requires an extra instruction
for the zero extend, but this is just one instruction.  It is really only
MIPS64 and RV64 that pay the penalty here, as both need two shifts for the
zero-extend.

GCC does have a value-range-propagation (VRP) pass, but this is on the high
level IL, where we have a 32-bit value whose range is computed to be
between 0 and 1000, then this value feeds into a zero-extend to 64-bits,
and the fact that we know the value range does not allow us to optimize
away the zero-extend.  It isn't until we lower to the target dependent IL
that we know that the value is stored sign-extended in a 64-bit reg, and we
can optimize away a zero-extension if the sign bit isn't set, but we don't
have the value-range info here; it has already been invalidated by other
optimization passes and thrown away.  Maybe if we used the value range info
to convert the zero-extend to a either-zero-or-sign-extend-is-ok operation,
then when we lower we could see that the extension could be optimized
away.  But I don't think we have a way to represent this in the high level
IL currently, and extending the high level IL would require modifying a lot
of optimization passes to be able to propagate through the new type of
extension to reach the lowering pass.  There may also be other better ways
to fix this.  Anyways, I think this is fixable if someone had enough time
to work on GCC to add the missing optimization.  We just don't have enough
people trying to optimize GCC RISC-V support yet, so it could be a while
before someone looks at this.

Jim

Serentty, 2019-07-10:
I would like to point out that there are programming languages (such as Rust) where array indices are _always_ unsigned, and the programmer has no choice in the matter. However, in Rust, array indices are also always the same size as a pointer, so they wouldn't usually need to be extended.

anon17, 2019-08-05:
If calling convention is a problem, compilers can provide an override and use, say, fastcall convention. Not sure if there's an overall good solution: if a system prefers some sort of integers, it's bad for other integers. But seriously, use of unsigned integers for positive numbers was a mistake. For example, unsigned integer overflow resulted in BlueBorne vulnerability and it can't be detected automatically, because unsigned integer overflow is legal in C, but overflow doesn't make sense for proper numbers like array indexes.

jnk0le, 2019-08-05:
> 
> 
> But seriously, use of unsigned integers for positive numbers was a mistake. For example, unsigned integer overflow resulted in BlueBorne vulnerability and it can't be detected automatically, because unsigned integer overflow is legal in C.

signed CVE's are also not uncommon:
https://nvd.nist.gov/vuln/detail/CVE-2018-19788

Having two possible overflows instead of one doesn't seem to increase security.

Serentty, 2019-08-06:
> But seriously, use of unsigned integers for positive numbers was a mistake. For example, unsigned integer overflow resulted in BlueBorne vulnerability and it can't be detected automatically, because unsigned integer overflow is legal in C.

This argument makes little sense to me. Sure, unsigned overflow is legal in C, but signed overflow is _undefined behaviour_ which is even _worse_. Sure, won't overflow by going below zero, but if you have an index that's less than zero and you access an array with it, you're still in for a very bad time.

anon17, 2019-08-06:
Undefined behavior can unconditionally disallowed by tools. You're going to have negative indexes anyway as programs tend to use `-1` as sentinel value.

Serentty, 2019-08-06:
> Undefined behavior can unconditionally disallowed by tools.

You can't have a lint for all undefined behaviour in C, as that would require solving the halting problem. Even if that weren't the case, whether or not overflow occurs can be determined by user input that isn't known at compile-time. The only way to statically verify that undefined behaviour doesn't occur is with a different language design, such as what Rust does (and even then sometimes you need to write code in ``unsafe`` blocks that the compiler can't statically verify for you).

> You're going to have negative indexes anyway as programs tend to use -1 as sentinel value.

That's traditional, yes, but many languages these days are going with option types as the idiomatic way to indicate such things instead.

Also, I would like to point out the great convenience there is in having array indices and pointers have the same range of possible values. It means that there are no artificial limitations on the sizes of things other than the size of the address space itself.

anon17, 2019-08-12:
>The only way to statically verify that undefined behaviour doesn't occur

You can't even try, because unsigned integer overflow is legal by design and has legitimate uses like bit manipulation. And rust [does absolutely nothing there](http://ideone.com/IPbm06) of course.

>It means that there are no artificial limitations on the sizes of things other than the size of the address space itself.

It's of little utility if at all. And even if you can come up with a use case, you can trivially address it with software design much like the option type.

harrysarson, 2019-08-12:
If I may: when compiled in debug mode [rust panics on unsinged integer overflow](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=5e68b5bb0e0afd4259aa0de060cf4049).

Serentty, 2019-08-21:
What you're saying still makes no sense. You're talking about how unsigned overflow being legal is dangerous and can lead to bugs, and therefore most numbers should be stored as signed instead. But signed overflow is undefined behaviour in C, which means that there are even more opportunities for bugs when you use them, as the standard allows pretty much anything to happen.

> You can't even try, because unsigned integer overflow is legal by design and has legitimate uses like bit manipulation. And rust does absolutely nothing there of course.

It seems like you interpreted what I said as ensuring that overflow doesn't occur. That's not what I meant. I said undefined behaviour, and that's what I meant. Rust does indeed 100% guarantee that you can not cause undefined behaviour through overflow, and it does that by defining what happens when integers overflow. The legitimate uses that you speak of such as bit fiddling benefit from this, as in C such code often relies on undefined behaviour.